polarofficial/polar-ble-sdk #1: Cannot get androidBleTestrApp to build

KennethEvans, 2019-03-21:
I extracted the  androidBleTestrApp from the ZIP download as a single project.  I also copied libs from polar-sdk-android to the project.  On opening the project in Android Studio, it asks to configure the Gradle wrapper.  (There is no gradle directory in the ZIP download.)

I then get this error:
C:\$Recycle.Bin\S-1-5-18
<a href="openFile:C:\AndroidStudioProjects\PolarBleSdkTestApp\app\build.gradle">Open File</a>	

This is likely owing to the
    copy {
        from("$projectDir/../../../../polar-sdk-android/libs")
        into("$projectDir/libs")
    }
since the path is not right.  (Also it should not be needed if I copy the libs directory manually.) 

I have not been able to get around it by commenting out these lines or the preBuild.dependsOn copySdk.

If I create a project polar-sdk-android project with the libs, and change the "from" line to be correct for that project:

from("$projectDir/../polar-sdk-android/libs")

then I get 
 
ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Failed to transform file 'polar-ble-sdk.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Failed to transform file 'polar-protobuf-release.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugAndroidTest/compileClasspath': Failed to transform file 'polar-ble-sdk.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugAndroidTest/compileClasspath': Failed to transform file 'polar-protobuf-release.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugUnitTest/compileClasspath': Failed to transform file 'polar-ble-sdk.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugUnitTest/compileClasspath': Failed to transform file 'polar-protobuf-release.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@release/compileClasspath': Failed to transform file 'polar-ble-sdk.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@release/compileClasspath': Failed to transform file 'polar-protobuf-release.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@releaseUnitTest/compileClasspath': Failed to transform file 'polar-ble-sdk.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@releaseUnitTest/compileClasspath': Failed to transform file 'polar-protobuf-release.aar' to match attributes {artifactType=android-exploded-aar} using transform ExtractAarTransform
Show Details
Affected Modules: app

So there seems to be something wrong with the files in lib that I cannot figure out.

In general there needs to be instructions to build the Android example alone without extracting the whole Github tree.  The ones in README.md did not help me.

Thanks.

KennethEvans, 2019-03-21:
I fixed it by manually copying the libs to the app folder, not the project folder.  (I'm used to putting libraries in the project folder.  I see this is not a library module, though.)  Apparently $projectDir is the app folder.  I also changed the path to 

from("$projectDir/../../polar-sdk-android/libs")

Hopefully that is now right.  I would like to get rid of the copy step entirely.  Have to figure that out.

So it now runs, but it doesn't do anything.  Pressing any of the options like Broadcast, Connect, etc. doesn't do anything.  Looking at the code, it looks like all it does is Log.  Am I missing something?

erkki-silvola, 2019-03-22:
README should have pointed out that example app + demo apps depend on the directory structure. Originally this was done so that the compiled libraries would not be in many places only in one directory.  Regarding your build error can you run gradle build with stacktrace enabled ?

erkki-silvola, 2019-03-22:
If you want the copy step away, remove the 'preBuild.dependsOn copySdk' and copy task itself. Manually copy 'polar-protobuf-release.aar' and 'polar-ble-sdk.aar' to app/libs folder. About the example app, it is just to show different features available in the sdk without any fancy UI, so it only prints messages to logcat. What is the feature you are interested in ?

KennethEvans, 2019-03-22:
Thanks for the reply.  I did comment those out,  and   it still gave me the Recycle Bin error. It must have cached them somehow. In any event I know how to make it work now.

As to what I want,  it was just to explore and see what the SDK can do. I expected the example to be more of an example. I see there are demos. I will look  at those. 

The one feature I wanted to look at is the ECG with the H10. I don't see how it can do that using BLE, at least using the Heart Characteristic.

KennethEvans, 2019-04-29:
I could probably mark this as closed.  Having encountered the problem, I take responsibility for not fixing it correctly at first, but don't think I should have been able to anticipate that the copyTask would not work for my file structure.

Later I found out using the path in copySdk as downloaded caused even more problems than I thought.  It turns out it copied a $GetCurrent directory that was huge.  (I now know $GetCurrent is a backup for Windows updates.)  It got into my Git repository as a huge blob before I realized that it shouldn't be there and removed it.  It caused Git to not work well for some tasks, and it took me a while to get rid of it from the early commits.  {I have a working knowledge of Git but am not an expert.)

The bottom line is that using copyTask can be a booby trap.  I think the problem would be minimized if it only looked for AAR files.  This is what I am using now.  (Note that this is for my particular directory structure and is currently using the branch.)

```
task copySdk {
    copy {
        from("$projectDir/../../polar-sdk-android-no-filter/libs")
        into("$projectDir/libs")
        include '*.aar'
    }
}
```
I'm also not a Gradle expert, so maybe there is a better implementation.

closed 2019-05-06 by erkki-silvola