polarofficial/polar-ble-sdk #2: Typo in the API documentation

seriousManual, 2019-03-22:
Since I did not find another way to contact somebody responsible for the API documentation, I'm posting this here:

The (otherwise great) API documentation says that the list of daily activities can be retrieved by the URL `GET /v3/users/{user-id}/activity-transactions/{transaction-id}` yielding a response in the form:
````
{
  "activity-logs": [
    "https://www.polaraccesslink.com/v3/users/12/activity-transactions/34/activities/56",
    "https://www.polaraccesslink.com/v3/users/12/activity-transactions/34/activities/120"
  ]
}
````
Other than stated in the documentation the key for the activities URLs is not `activity-logs` but `activity-log`:

[image]

erkki-silvola, 2019-04-03:
I will forward this issue onwards.

seriousManual, 2019-04-03:
Can you tell me how the developer support for API integrations can be reached?

vmhirvinen, 2019-07-03:
Good catch! API Documentation should now reflect reality better for daily activities.

There is support email in top of API documentation.

closed 2019-07-03 by vmhirvinen