polarofficial/polar-ble-sdk #3: No ECG in polarsdkecghrdemo

Je-Ho, 2019-03-23:
I'm trying the demo app polarsdkecghrdemo with a Polar H10 - connection works, HR is shown but ECG is not shown. Logcat does not show ECG either, only HR. Should it work with an H10? 

[image]

erkki-silvola, 2019-03-24:
You most likely don't have the version from H10 which supports ecg streaming.

Je-Ho, 2019-03-24:
Thank you, I bought the sensor a few weeks ago and there is no firmware update higher than my installed version 2.1.9, so I didn't think about that 😉. Will it work with a future firmware update or is there a hardware update as well? Thanks again for making the SDK available!

erkki-silvola, 2019-03-25:
There will be update later for H10.

KennethEvans, 2019-03-26:
I have the same issue,  The PolarSdk-Exercise-Demo doesn't do anything either, just sits there.  Doesn't even display the HR.  After looking around I am not finding any real documentation on the SDK, apart from the Javadoc, which is pretty terse.  There is a Gatt spec and a DOCX that says the H10 does ECG with a sample rate of 130 Hz.  These aren't much help in understanding the SDK.

I also have version 2.1.9 on the H10.  What I read is that Polar beat will notify me when there is an update.  I am not seeing much use for the SDK as is.  Is this update coming soon?

erkki-silvola, 2019-03-27:
The update will be available soon.

Je-Ho, 2019-04-16:
Nice, the update is working, thanks!

closed 2019-04-16 by Je-Ho

KennethEvans, 2019-04-25:
It's been 8 days since the response above, indicating the update is out.  However, I have not got an update notice for my H10.  My understanding is it will show up in Polar Beat.  Am I missing something?

Is there a manual way to get it?

KennethEvans, 2019-04-25:
Apparently it came out yesterday (April 23). I just got it and am now getting ECG.