polarofficial/polar-ble-sdk #5: polarblesdk not working with latest xcode/swift updates

Roberdan, 2019-03-26:
hi, I just updated Xcode and I'm having this error message when trying to compile my project: 
"Module compiled with Swift 4.2.1 cannot be imported by the Swift 5.0 compiler: /Users/roberdan/Desktop/MirrorHR2.0/PolarBleSdk.framework/Modules/PolarBleSdk.swiftmodule/x86_64.swiftmodule", any help?
thanks a lot, RD

erkki-silvola, 2019-03-27:
I will check this next week's tuesday.

Roberdan, 2019-03-28:
any chance to recompile it earlier? we have a deadline and we are blocked :( thx

erkki-silvola, 2019-03-29:
Sorry I cannot as I am on vacation still, and I don't have my MacBook with me.

erkki-silvola, 2019-04-03:
Now there is swift 5 compatible sdk available. Note I still need to check demos and re-generate docs and testing before 1.1.0 release tag.

closed 2019-05-06 by erkki-silvola