polarofficial/polar-ble-sdk #6: PolarSDK-ECG-HR-Demo with HR and RR Plotting

KennethEvans, 2019-03-30:
I modified the PolarSDK-ECG-HR-Demo app to add an Activity that plots HR and RR as a strip chart.  Screenshot attached.  I can send you the changes if you are interested, no obligation.  It seems like it would be useful as a demo to others.

[image]

erkki-silvola, 2019-03-31:
You can send the changes. Demos are originally done by a trainee so I have not personally checked them, but any improvements are welcome.

hepcatjk, 2019-04-01:
Can we reproduce this demo on the computer too somehow with the SDK? using BLE and python/c# or C++ or something?

erkki-silvola, 2019-04-01:
Of course it is possible. But in that case you need to implement the gatt client yourself. Check the gatt specification for details.

pedromi, 2019-04-02:
Hi KennethEvans! Can you give me more information / send me of your activity to test? Many thanks

KennethEvans, 2019-04-02:
I did send what I have done.  It says you can reply to the notification I got from this topic, and that is what I did.  Since no mention has been made of its being received, I assume it did not get to where it needed to be.

Let me try attaching it here:
[PolarSDK-ECG-HR-Demo-Modified.zip](https://github.com/polarofficial/polar-ble-sdk/files/3035004/PolarSDK-ECG-HR-Demo-Modified.zip)

This is a ZIP of the main directory.  The only changes I made were to the java files and ones in res/layout and res/values/strings.xml.  The screenshot I sent is from the added HRActivity.  I did also make some minor changes to ECGActivitiy and its layout to add some more information.  You may or may not want to keep that.  (I personally can't do anything significant with ECG until my H10 gets updated.)

erkki-silvola, 2019-04-03:
I updated egg demo with @KennethEvans changes

closed 2019-05-06 by erkki-silvola