polarofficial/polar-ble-sdk #8: not working in background?

Roberdan, 2019-04-04:
hi there,
I'm not able to have data from the HR sensor when the app is in background (background mode in the app is ok and other sensors are working fine), do you have any insight/suggestion?thanks, Rd

erkki-silvola, 2019-04-04:
I just tested with polarBleSdkTestApp and it worked ok. What do you mean by "other sensors" ?

Roberdan, 2019-04-04:
I see, the testapp works, for sure I'm doing something wrong but I do still have issues :(...what are the exact steps just for being sure it should work? thanks in advance

erkki-silvola, 2019-04-05:
Is it so that immediately when app goes background hr notifications stops?

Roberdan, 2019-04-05:
just few secs later, yes

erkki-silvola, 2019-04-05:
Have you tried switching bt off/on or restarting the phone? One reason could be that os kills the app because of memory pressure etc. What is the phone/tablet model you are using?

Roberdan, 2019-04-07:
tried, still same issue. Shall I call the startrecording/stoprecording functions? I was trying to use just the broadcasting. thx

erkki-silvola, 2019-04-07:
Broadcast listening needs to have service in the scanning filter while on background which means broadcast hr listening only works when sensor is not connected somewhere else. Except H10 with dual connection enabled.

closed 2019-05-06 by erkki-silvola