polarofficial/polar-ble-sdk #10: Trouble with PPI and PPG Feature for OH1

Markeey, 2019-04-05:
Hi,
i am trying to test the PPI and the PPG Feature of the OH1 sensor connected to your Android example application. The demo application works well in terms of printing current heartrate values to the log console:

**Example Message (HR is working):**
2019-04-05 23:18:25.171 32086-32086/polar.com.androidblesdk D/MainActivity: HR value: 72 rrsMs: [] rr: [] contact: false,false

**But if I hit the PPG_STREAM or the PPI_STREAM button in the APP i only see the following line in log:**
2019-04-05 23:20:10.944 32086-32086/polar.com.androidblesdk E/MainActivity: null

Furthermore while debugging i can clearly see that the ppiFeatureReady and the ppgFeatureReady Methods in the MainActivity.java class are never be called.

Maybe you got an advice what I Need to do to make this work.

Thank you for your reply.

Best regards,

Mark

erkki-silvola, 2019-04-06:
Hi, 

The reason is simple you don't have the version from OH1 which supports ppi and ppg streaming. Check fwInformationReceived callback for the version. OH1 update will be available soon.

Markeey, 2019-04-08:
Thanks for your fast Reply! I am really Looking Forward to this upgrade :)

KennethEvans, 2019-04-25:
Ok, I got an OH1, and it is 2.0.8, so I can get PPG and PPI data.  The problem is I don't know what most of the output means (ppg0, ppg1, ppg2, ambient, blocker, error estimate, ...).  Where would I find this information?  I didn't see it in the documentation, but I could have missed it.

The real bummer is that it doesn't output RR values.  (I know there is no R wave in a PPG graph but it could output the time between pulses (PPI?) as the RR values.  It has to know that to calculate a HR.)  My CorSense, which is a PPG, does this.  It would make it usable with non-Polar apps.

erkki-silvola, 2019-04-25:
ppg = Photoplethysmography. Numbers 0,1,2 is the channel number of the ppg signal. This is shortly documented in the docs. 
ambient = Integer value that is used to cancel ambient light while the measurement is being made from the skin.

*** I know there is no R wave in a PPG**
- That's right there is no real R in PPG.

For now you need to use separate PPI measurement.

KennethEvans, 2019-04-25:
> "shortly: documented

Yes, if shortly means terse.

The documentation has something to say about the services and characteristics.  I find nothing about the definitions of parameters such as ppg0, ppg1, ppg2, etc.  For ECG data the equivalent information is pretty well known  (e.g. RR), and can be readily found.  For PPG I find papers on multi-channel PPG, which is what I assume the ppg0, etc. values are.  However, there is no standard on how many channels to use, and I do not find definitions of other things such as how the pppg0, etc. numbers are defined.  Similarly the definitions of things like ambient, blocker, error estimate, etc. do not appear to be standard or where I can find them.  These, in fact, seem to be specific to Polar's implementation of both the hardware (6 LEDs, for example) and the OH1 software that determines what to send.  I don't see why Polar would want to keep these definitions a secret (even though they might want to keep their implementation a secret).  I need to know them to interpret the data.

The important things for PPI are the timestamp which is a member of PolarOhrPPIData and the hr and ppi which are members of PolarOhrPPISample, of which there are more than one for a given PolarOhrPPIData.  Unlike pg0, I can guess what timestamp, hr, and ppi mean, but I still don't understand the SDK implementation.

There is only one timestamp in the PolarOhrPPIData, but several samples.  How do you determine at what time the samples occurred?  Actually I don't understand why there are multiple samples.  The PPI Frame type has hr(1B), ppi (2B), error estimate (2B), and flags (1B).  There is no timestamp, and I assume the SDK makes the timestamp be the time the data arrived.  (This is different from an HR characteristic which has 1 hr and multiple rr.)

So a question that seems to be an SDK thing and that you should be able to answer is why there are multiple samples and how are they related to the timestamp?  I am currently stuck on this.

Actually I guess I also really need to know the definition of the timestamp (units and base time), even though I can probably determine it by trial and error.

I hope I am not being difficult (or dense).

Thanks.

KennethEvans, 2019-04-25:
Later: Looking at androidBleSdkTestApp I am seeing the timestamp is 0 and there is only one sample per call.

If there is only one sample per call, then I can take the time the data arrived, and use that one value, no problem.  However, I worry I am missing something.

KennethEvans, 2019-04-25:
Ok, a little trial and error goes a long way when you don't have good documentation.  This is what I discovered using the OH1:

1. The timestamp in the PolarOhrPPIData is always zero.
2. the hr in the samples is also always zero.
3. There are typically many samples in the PolarOhrPPIData.  These correspond to the array of RR values from the PolarHrData.

Thus for the OH1 you have to use both the HR feature (for hr) and the PPI feature (for the rr, a.k.a. ppi).

I got this working in my HR-Compare Demo.  I have reason to believe this app is working (but it needs to be tested more over time to be sure).  It works as expected for the H10, and works for my CorSense, which is a PPG device (except that I think their RR intervals are consistently low and it misses some as indicated by the fact their RR values don't add up to the elapsed time.)

What I observe for the OH1 are:

- The PPI intervals do add to the elapsed time (a sanity check).
- It takes a long time to start up the PPI.  It reports HR much earlier, but incorrectly.
- While starting up it reports a zero heart rate.
- It intermittently reports a zero heart rate at other times (verified by another app) and has drops in the hr compared to the H10.
- The HRV pattern seems to be delayed compared to the H10.

I am attaching a screenshot and also a similar screenshot for the CorSense.  You can draw your own conclusions, but IMHO the H10 is much better if you are interested in getting than a rough HR.  Also IMHO the CorSense is a better implementation of PPG.  It works in normal apps and does not require an SDK and all that extra, hard-to-obtain knowledge about PPI and PPG.

On the positive side the SDK seems to be well implemented and has been very robust so far in my experience.  It's certainly easier than doing the Bluetooth yourself.  Being able to connect to two devices at once is neat.

In any event, this app, for which the SDK is doing the heavy lifting, seems to be a good way to compare devices quantitatively.

H10 vs. OH1

[image]

H10 vs. Corsense

[image]

erkki-silvola, 2019-04-29:
* The timestamp in the PolarOhrPPIData is always zero.
   This is due the internal logic of OH1, and it cannot produce accurate timestamps.

* the hr in the samples is also always zero.
  Again there is some internal logic in OH1 affecting this.

*  It takes a long time to start up the PPI. It reports HR much earlier, but incorrectly.
  Again there is some internal logic in OH1 affecting this.

*  While starting up it reports a zero heart rate.
  Again there is some internal logic in OH1 affecting this.

I'll try to improve the documentation, so that people can avoid the same "trial and error" approach.

KennethEvans, 2019-04-29:
Thanks for the information.  I also noted that the HR may go to zero for short periods after the start up period.  This seems to stop happening after a while.

As to the documentation, these things appear to be device specific.  The SDK itself seems to honor the spec.

On the subject of timestamps, I have found the ECG data do have a valid timestamp.  (This is for the H10 though.)  The problem is that the specification in the documentation does not specify the epoch, so it cannot be used to get the actual time.  Again, by experimenting: a few days ago it appeared to be Apr 26 18:52:22 GMT 2000 but now I see it is Apr 28 14:23:27 GMT 2000.

It would be nice if device details like these (PPG, Timestamps, etc.)  were available.  I would assume that is not your responsibility, though.

Thanks,

erkki-silvola, 2019-05-02:
* On the subject of timestamps, I have found the ECG data do have a valid timestamp. (This is for the H10 though.) The problem is that the specification in the documentation does not specify the epoch, so it cannot be used to get the actual time. Again, by experimenting: a few days ago it appeared to be Apr 26 18:52:22 GMT 2000 but now I see it is Apr 28 14:23:27 GMT 2000.
For H10 "calendar" starts from 2000. There is a way to set time on device using "setLocalTime" API function though it requires to enable FEATURE_POLAR_FILE_TRANSFER.

* It would be nice if device details like these (PPG, Timestamps, etc.) were available. I would assume that is not your responsibility, though.
This is a "gray" zone. Partially it is on my responsibility and partially not. I will try to improve documentation to avoid misunderstanding.

I hope these answers helped.

closed 2019-05-06 by erkki-silvola