polarofficial/polar-ble-sdk #9: Use with Non-Polar Devices

KennethEvans, 2019-04-05:
As is, the SDK only works with Polar devices; i.e. you have to enter a serial number, and it filters out non-Polar device names.  I would like to use it with other devices, many of which support the BLE Heart Rate Service.

On looking through the code it looks like the access to Android Bluetooth is through the BleDeviceSession and its descendants.  In particular the basic Android BLE interface is in the BluetoothDevice and BlutoothGatt members of BleDeviceDeviceSessionImpl.  However, since these are private I cannot access them.

I could, for example, connect to other devices by extending BleDeviceDeviceSessionImpl and adding another connect method.  I could also do other things like scan for devices or get more device info.

I don't claim to understand the SDK code completely, but it looks like the identifier doesn't really have to be a Polar serial number, at least for non-Polar-specific features.  It seems to be promarily used for connecting.  Presumably there would be a Exception generated if one tried to use those features and they are not available (or else nothing would happen as with the ECG for non-updated H10's).

So far my experience is that the SDK is robust, and I am sure you have put more effort and expertise into it than whatever I might have done myself.  However, I don't want to use it to write apps that are limited to Polar (when that is not necessary).  It's not that I don't like Polar.  It's that I have devices like the Corsense finger sensor that Polar does not make.

So is this intentional?  Cast in stone? What are your thoughts?

Thanks.

erkki-silvola, 2019-04-06:
The SDK was intentionally non obfuscated as there are no secrets inside it (source code distribution is scheduled to later point in time. Meanwhile people are free to discover the android aar package, though non documented). And yes the SDK was intentionally made to work only with Polar sensors. If you want to remove polar specific limitation you only need to remove it from BDBleApiImpl class. Basically it means changing polar device id identifier to bt address in the api. I will consider removing this limitation from the SDK.

KennethEvans, 2019-04-06:
> you only need to remove it from BDBleApiImpl class

Yes. If its fields were protected rather than private, this should be straightforward to do. 

BTW the source is in the docs, and it can be decompiled in Android Studio.  As you said, it isn't secret.  Thanks.

erkki-silvola, 2019-04-08:
I thought you were going to re compile the sdk without polar devices restriction, but seems this is not the case. I will change the default implementation BDBleApiImpl class so that one can extend this class and I will add to example app extension class to show how to remove polar specific filters.

KennethEvans, 2019-04-08:
Cool. Thanks.

Much easier than trying to maintain a parallel sdk ;-)

erkki-silvola, 2019-04-09:
I pushed new branch ( no_polar_filter_api_extension ), where is for extended implementation available, check polarBleSdkTestApp for that, iOS needs the same update, which is in working progress.

KennethEvans, 2019-04-10:
Ok, I used the defaultImplementationWithoutPolarFilter.  (I didn't see the comment.)  It is a little hard to tell what it is doing since the only result is messages in Logcat.  It does seem to be finding the Corsense device, along with all the other BlueTooth devices (BLE or not) in range.

I have been able to get it to connect to the H10 and show HR and RR messages.  I have not been able to do that with the Corsense at all.  (I have no trouble connecting and getting data from the Corsense with other apps.)

With the H10 I am not sure what causes that to happen, and I can't consistently make it happen.  It definitely does not happen after tapping the CONNECT button, at least not very often.  I have had better luck  with AUTO_CONNECT_TOGGLE (why toggle?), but more often than not it does not happen.

It looks like it will take some time to figure out what this app is supposed to be doing.

Maybe it would be better to try making the Polar-ECG-HR-Demo app work with this sdk.  At least there I know what it is supposed to do.

Thanks.

erkki-silvola, 2019-04-10:
Hi, I changed the implementation to have only the defaultImplementation and I added setPolarFilter api to switch off/on polar filter. Notice now all api functions should work with either bt address or polar device id. I just tested the main functionality with 3rd party sensor. If you don't know the bt address of Corsense use either autoConnect or use first searchForPolarDevice to find out bt address.  If your Corsense is advertising you should see this in logcat "new device allocated name: xxxx".

KennethEvans, 2019-04-10:
Cool. How do you use the setPolarFilter API? I am having trouble with the androidBleTestApp. Can I use the new stuff anywhere by just replacing the two AAR files in libs?

I do know the address of the Corsense.  By this I assume you mean the full display name, not the Mac address. Can you still use just the serial number for Polar devices?

erkki-silvola, 2019-04-11:
Yes you can replace the aar files, though notice that android studio poorly recognise changes made in aar files, although compilation should work ok. Check androidBleTestApp after api initialisation I just call setPolarFilter(false) to disable polar filter. With bt address I mean Mac address. All api functions should now work with either mac address or polar device id.

KennethEvans, 2019-04-11:
Thanks.  Using the MAC address worked, and I connected to my Corsense.  I did not do anything to androidBleSdkTestApp except change the DEVICE_ID.

AAR Problem: File | Invalidate Caches/Restart seemed to work.

KennethEvans, 2019-04-12:
And I was also able to get the PolarSDK-ECG-HR-Demo working with the Corsense with little trouble after replacing the AAR files and adding the api.setPolarFilter(false).

Interestingly, this device uses the first RR value to send a code to the device.  The RR value is < 10 so it is unlikely a real RR interval.  However, if you don't do something special with the RR values (like delete the first one), then the low values would show in the plot.  I didn't do anything special in the PolarSDK-ECG-HR-Demo, but it is not showing in the plot.  Are you doing something special in the SDK?

Thanks.

KennethEvans, 2019-04-12:
On playing it with it a bit, I see there is a lot of information that isn't reachable, like most of the stuff in the Device Information Service.

Or maybe I missed it?

erkki-silvola, 2019-04-12:
From DIS only software revision is read. There is no special handling regarding RR values inside SDK.

KennethEvans, 2019-04-12:
> There is no special handling regarding RR values inside SDK.

Something is happening.  The device __is__ sending out low values <~10, and they aren't coming into the callback.  I see them with other software, and the people responsible (Elite HRV) verify they are there (and presumably used in their app).  (It's an annoyance and violates the BLE specs IMHO.)  I expected to have to remove those values when plotting with the SDK, but I don't.

It doesn't seem to be a problem.  I'm just curious.

BTW I was able to plot HR and RR from two devices simultaneously with the SDK.  (e.g. Compare H10 to CorSense).  That was relatively easy and something I've wanted to do for a long time.

Waiting for ECG to be enabled.

KennethEvans, 2019-04-23:
The CorSense is not really sending low values.  They were Energy Expended (EE), which my code was not handling correctly  (and unnoticed for years since it worked when EE is not included).  The bad code was probably originally derived from a demo, also possibly used by the other software I tested.  The SDK appears to be doing it correctly and caused me to examine the logic more closely.  I apologize.

closed 2019-05-06 by erkki-silvola