polarofficial/polar-ble-sdk #11: Background feature explanation.

RAN3000, 2019-04-12:
Hi, 
I see that there is a `api.backgroundEntered();` method.
I'd like to know a little bit of details on how it works if possible.
On Android there are many ways to achieve background functionalities, which one did you use?
When is it going to be killed by the system?
Does it run on the main thread?
In the callback with the HR results for example can we do arbitrary things, or do we have restricted resources?
As I understand iOS has a background functionality too, can you also expand on that?

I did experiment a bit with it, but I'd like to know a little more from you.
I couldn't find any information on the documentation.

Thanks.

erkki-silvola, 2019-04-15:
Hi,

**On Android there are many ways to achieve background functionalities, which one did you use?**
  * The api functions backgroudEntered() and foregroundEntered() only enable / disable scan filter, as you need to specify a service which to scan while on background, note this applies only from android API 26 onwards. 

**When is it going to be killed by the system?
Does it run on the main thread?**
  * Application  developer decides the Context, on the example app and demos they are run on Main   Thread, but if you want to run it on Android Service that is fine too.

**In the callback with the HR results for example can we do arbitrary things, or do we have restricted resources?**
Basically there is no restrictions regarding hr received callback, but if you throw exception hr notifications receiving stops

**As I understand iOS has a background functionality too, can you also expand on that?**
For iOS the case is the same. Only scan filter is enabled / disabled when going to background and foreground.

RAN3000, 2019-04-15:
Thank you for your explanation.

I still need some clarifications if possible.

I'm currently playing with the [androidBleSdkTestApp](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) in this repo.
If I do the following actions: connect to a device, press the home button, lock the device; the app keeps the connection with the H10 open (I see the hr stream in the logcat and I also write it to a file), if I disconnect the device it starts scanning again (around every 5sec) and eventually reconnects appropriately. This happens until I reopen the app and disconnect from the device.

My question is: is it just a coincidence that my device does not kill the app? Or is this the behavior you wanted? 
If not, do you have any suggestions on how to implement this behavior? Maybe a foreground service?
(if I comment `api.backgroundEntered();` in the `onPause` this behavior happens anyways)

Also, as I understand, this behavior is supported in iOS by enabling in project target settings `Background Modes` and adding `Uses Bluetooth LE accessories`. Is this correct?

erkki-silvola, 2019-04-15:
**My question is: is it just a coincidence that my device does not kill the app? Or is this the behavior you wanted?**
  * The behaviour is wanted, as by default automatic reconnection is enabled until you call disconnect from device or disable automatic reconnection totally. For long running background apps it depends on phone manufacturer, on some devices you need explicitly enable background mode so that it continues to run on background. Generally for long running tasks I would recommend Android Service to be used.

**(if I comment api.backgroundEntered(); in the onPause this behavior happens anyways)**
  * This depends if the scanning works without service filter on background, should work at least prior API 26. Though phone manufacturers may have tuned this on some devices.

**Also, as I understand, this behavior is supported in iOS by enabling in project target settings Background Modes and adding Uses Bluetooth LE accessories. Is this correct?**
  * This is correct.

RAN3000, 2019-04-15:
I appreciate your taking the time to answer my questions.
Thank you!

closed 2019-04-15 by RAN3000