polarofficial/polar-ble-sdk #12: When will new firmware of H10 supporting ECG be ready to be updated?

erikguo, 2019-04-15:
We are testing the ECG feature of H10.  Then we found we should wait for the new firmware. So , when will the new one be ready?

Thank you very much

erkki-silvola, 2019-04-15:
Update will be available soon.

erikguo, 2019-04-16:
Thank you for your quick reply.

How about the following problem?

And we find the version of firmware H10 is v1.1.8 now. When we open the Polar Beat, we didn't get the update information. So we cannot update to the latest version v2.1.9. Could recommend some solution for us?

We are in China.
Polar Beat v3.2.0 is downloaded from Apple App Store (China Area).
iOS is 12.2. iPhone 6

erkki-silvola, 2019-04-16:
Currently H10 update is "blocked". It will be enabled soon and you are able to update to 3.x version.

closed 2019-05-06 by erkki-silvola