polarofficial/polar-ble-sdk #14: PolarBleApiCallbackAdapter

KennethEvans, 2019-04-29:
This is a feature request.  It would be nice to have a PolarBleApiCallbackAdapter that works similar to a MouseAdapter.  Then you wouldn't have to clutter your code with api callbacks in which you are not interested.

It is trivial.  I am including what I did.

[PolarBleApiCallbackAdapter.txt](https://github.com/polarofficial/polar-ble-sdk/files/3128242/PolarBleApiCallbackAdapter.txt)

erkki-silvola, 2019-05-02:
True I will do this for next release, which will be end of next week.

KennethEvans, 2019-05-02:
I am probably near the end of playing with the SDK.  I now have a fairly full-fledged ECG app, which was my major interest.  It is very impressive that a strap can do that.  I have been using Polar since they were wrist watches that showed the current HR and some averages.  This is quite an advance.

I have put my stuff at https://github.com/KennethEvans/PolarSDK.  It has an MIT license.  Some of the apps there are little different than the SDK ones, probably with more logging.  They are in my version control because it is convenient for me.  There are only two with significant changes.

The version of PolarSDK-ECG-HR-Demo there has a better strip chart and a better RR placing algorithm than the previous one.

My project that is now Polar HR Compare was originally PolarSDK-ECG-HR-Compare-Demo.  The version as of the commit on 4/26/2019 11:07:28 AM my time labeled Minor changes to HR-Compare still had that name, Polar branding, and the original icon.  It could also be used as a demo if you would like.  It doesn't really have new features from the other except connecting to two devices at the same time, but it is a useful app IMHO.  You are welcome to include it if you like.

Both of these are using the branch and probably have the 

```
task copySdk {
    copy {
        from("$projectDir/../../polar-sdk-android-no-filter/libs")
        into("$projectDir/libs")
        include '*.aar'
    }
}
```
 pointing to my directory, not yours.  Perhaps at some point there will be releases so one could just update the libs directory manually as they become available.

I am very impressed with the SDK, and the changes in the branch made it more useful for me.  Thank you for doing it and for your time explaining.

closed 2019-05-07 by erkki-silvola