polarofficial/polar-ble-sdk #16: iOS framework with full bitcode support

kuradeveloper, 2019-05-04:
Hi, I was trying to archive my iOS project which uses the iOS version of the SDK and got the following linker error:

ld: bitcode bundle could not be generated because '/Users/...../project/polar-sdk-ios/PolarBleSdk.framework/PolarBleSdk' was built without full bitcode. All frameworks and dylibs for bitcode must be generated from Xcode Archive or Install build file '/Users/...../project/polar-sdk-ios/PolarBleSdk.framework/PolarBleSdk' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

My project builds and runs fine but archiving is a necessary step to publish to App Store. Is there anything I can do to correct this issue? If not, can Polar publish a new iOS framework with full bitcode support? Thanks

erkki-silvola, 2019-05-06:
Hi, can you try iOS sdk from develop branch, I did rebuild it with missing linker flags.

kuradeveloper, 2019-05-06:
The rebuild from the develop branch solved the problem. Thanks for the help

closed 2019-05-06 by kuradeveloper