polarofficial/polar-ble-sdk #17: Calories burned?

linktohack, 2019-05-07:
Hello,

I would like to develop a dashboard (say to display on a TV) that show my current heart rate + calories burned in my session.

1) Is it possible with the SDK?
2) If I am going to display the information from more than one device, can I use the SDK or have to use OPEN ACCESSLINK API?
3) In this case can I get realtime access to (current heart rate + calories), something like [1: the doc] but via a web hook?

[1: the doc]: https://www.polar.com/accesslink-api/#get-activity-summary

erkki-silvola, 2019-05-07:
I would like to develop a dashboard (say to display on a TV) that show my current heart rate + calories burned in my session.
     **What is the platform used?**

linktohack, 2019-05-07:
iPad then mirror to a TV via AirPlay (if not possible then a tablet Android then mirror to a TV via ChromeCast)

Thanks for your quick reply @erkki-silvola

erkki-silvola, 2019-05-07:
1. Is it possible with the SDK?
**For calories no. Polar sensors does not support energy expended directly from sensor. Though there are many open source implementations for calories consumed based on hr, gender etc... If you don't have your own calories calculation.**

2. If I am going to display the information from more than one device, can I use the SDK or have to use OPEN ACCESSLINK API?
**SDK does support multiple connections.**

3. In this case can I get realtime access to (current heart rate + calories), something like 1: the doc but via a web hook?
**No.**

jussivirkkala, 2019-05-07:
I noticed that both H10 and OH1 contains HR as two last bytes of BLE advertisement packet. Also 3 bit running index. Is there any documentation about these? You can listen more than 7 devices advertisement packets. 

With described GATT commands I think you can connect to multiple sensors. GATT documentation is working with e.g. https://github.com/aritchie/bluetoothle also on Windows 10.

linktohack, 2019-05-07:
Thanks again @erkki-silvola, please considered the issue closed.

closed 2019-05-07 by linktohack

jussivirkkala, 2020-09-30:
This HR information in advertisement package last byte seems visible also with Polar Unite (and I assume also Ignite) during workout if enabled.  When comparing to Polar Flow HR seems identical. Is the format so simple or is there something else we should be aware of, There is also many another constant bytes in advertisement.