polarofficial/polar-ble-sdk #19: PolarBleApiDeviceHrObserver getting PolarBleSdk.DeviceNotFound

Narragansett, 2019-05-17:
Have HR10 device. How to get RR data and PolarBleApiDeviceHrObserver to work?

I'm using the iOS Example app, hitting the buttons and get this in Xcode log ..
 Polar H10 514B382E HR BROADCAST: 53
 Polar H10 514B382E HR BROADCAST: 53
 Polar H10 514B382E HR BROADCAST: 53
 [BLE] commandState state:SCANNING action: CLIENT_REMOVED
 [BLE] commandState state:SCANNING action: EXIT
 [BLE] commandState state:IDLE action: ENTRY
 start ecg error: PolarBleSdk.DeviceNotFound
 recording start fail: PolarBleSdk.DeviceNotFound

What is causing DeviceNotFound if Broadcast is working? Is PolarBleApiDeviceHrObserver in a working state? Thank you,

Narragansett, 2019-05-17:
So if the Broadcast works for HR, then what causes this error for other functions? start ppg error: PolarBleSdk.DeviceNotFound .  What causes DeviceNotFound when the device is Bluetooth connected and Broadcast is working fine??

erkki-silvola, 2019-05-18:
Listening for broadcast hr is not a connected connection. Try auto connect to connect nearby polar device and after connected to sensor you will get callbacks like ecg feature ready etc.. then you will be able to start ecg stream etc...

Narragansett, 2019-05-18:
Yes! You are right again! The two connections are different. If I use auto-connect I do get callbacks to hrValueReceived. Its working now, ty!

closed 2019-05-18 by Narragansett