polarofficial/polar-ble-sdk #21: [HELP]Polar Bluetooth connection to Lenovo E7 tablet.

SamerDiab, 2019-06-05:
Hello, hope this finds you well. 

I have been experimenting with the polar OH1, the set up went smooth with a samsung galaxy note 5. But when i tried to run the app from a lenovo e7 tablet, it failed. This is from the logcat when i tried to connect to device. 
OH1 is updated to latest 2.0.10 firmware.

``` 
2019-06-06 01:12:29.753 8180-8180/polar.com.androidblesdk D/Polar BHE: version: 2.1.0
2019-06-06 01:12:29.757 8180-8180/polar.com.androidblesdk D/Polar BHE: BLE power: true
2019-06-06 01:12:29.797 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ADMIN_STOP_SCAN
2019-06-06 01:12:29.799 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: EXIT
2019-06-06 01:12:29.799 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ENTRY
2019-06-06 01:12:29.799 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ADMIN_START_SCAN
2019-06-06 01:12:29.799 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: EXIT
2019-06-06 01:12:29.800 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ENTRY
2019-06-06 01:12:29.871 8180-8180/polar.com.androidblesdk D/Polar BHE: BLE power: true
2019-06-06 01:12:29.889 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ADMIN_STOP_SCAN
2019-06-06 01:12:29.889 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: EXIT
2019-06-06 01:12:29.890 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ENTRY
2019-06-06 01:12:29.890 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ADMIN_START_SCAN
2019-06-06 01:12:29.890 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: EXIT
2019-06-06 01:12:29.891 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ENTRY
2019-06-06 01:12:30.594 8180-8180/polar.com.androidblesdk D/Polar BHE: bt ready
2019-06-06 01:12:30.596 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ADMIN_STOP_SCAN
2019-06-06 01:12:30.597 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: EXIT
2019-06-06 01:12:30.597 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ENTRY
2019-06-06 01:12:30.597 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ADMIN_START_SCAN
2019-06-06 01:12:30.597 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: EXIT
2019-06-06 01:12:30.597 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ENTRY
2019-06-06 01:12:33.621 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
2019-06-06 01:12:33.622 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: EXIT
2019-06-06 01:12:33.622 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:SCANNING action: ENTRY
2019-06-06 01:12:33.623 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/timestamps left: 0
2019-06-06 01:12:33.623 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/Scan started -->
2019-06-06 01:12:33.641 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/Scan started <--
2019-06-06 01:12:50.932 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:SCANNING action: ADMIN_STOP_SCAN
2019-06-06 01:12:50.932 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:SCANNING action: EXIT
2019-06-06 01:12:50.932 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/Stop scanning
2019-06-06 01:12:50.943 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ENTRY
2019-06-06 01:12:50.943 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: ADMIN_START_SCAN
2019-06-06 01:12:50.944 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:STOPPED action: EXIT
2019-06-06 01:12:50.945 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: ENTRY
2019-06-06 01:12:50.947 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:IDLE action: EXIT
2019-06-06 01:12:50.948 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/commandState state:SCANNING action: ENTRY
2019-06-06 01:12:50.949 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/timestamps left: 1
2019-06-06 01:12:50.949 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/Scan started -->
2019-06-06 01:12:50.971 8180-8180/polar.com.androidblesdk D/Polar BHE: BDScanCallback/Scan started <--

```

This is where the log stops.  
No exceptions where thrown, all permissions are acquired.
 
The tablet has bluetooth 4.0 according to specification sheet. 

If i enter the bluetooth page in device settings, i can see the device and pair to it.
I tried the following steps. 
Paired to the oh1 and tried to connect.
Unpaired to the oh1 and tried to connect.
Factory reset oh1 and tried to connect.
Tried to connect to 2.0.8 firmware before updating. 

**Nothing connected.**

Some help would be much appreciated. 

Thank you.

SamerDiab, 2019-06-05:
Update: Tried to install Polar Beat app on Lenovo E7 tablet, that failed too. :/
Any help?

erkki-silvola, 2019-06-06:
What is the connection method you trying to use, AutoConnect or ConnectToDevice?

SamerDiab, 2019-06-06:
@erkki-silvola thanks for replying. I tried both AutoConnect Or ConnectToDevice. I passed `-50, "180D", null` on auto connect. Both did not work. When i built the apps on other devices, they worked perfectly, so it is device specific for sure.  How would you like to go around this so that we can fix the issue?

erkki-silvola, 2019-06-06:
after looking at the logs it seems that the device does not find any device in scanning, what android version is in the tablet

SamerDiab, 2019-06-06:
@erkki-silvola  i enter the bluetooth page in device settings, i can see the device and pair to it. But not through code. The android version on it is 8.1.0, android oreo **go** version.

erkki-silvola, 2019-06-06:
can you try nRFConnect app from play store to check does it find anything?

SamerDiab, 2019-06-06:
@erkki-silvola i just installed the nrf app, and it found the polar oh1. Any data i should log?

erkki-silvola, 2019-06-06:
the problem might be the scanning api used in sdk, as it uses new api from lollipop onwards. The nrf app uses always the old api if i recall correctly

SamerDiab, 2019-06-06:
Any suggestions of what i can do? @erkki-silvola

erkki-silvola, 2019-06-06:
I need investigate this lenovo tablet, what is is exact model seen in adb ?

SamerDiab, 2019-06-06:
The tablet is a lenovo E7 tablet present in this link: https://www.lenovo.com/lb/en/tablets/android-tablets/tab-series/Lenovo-Tab-E7/p/ZZITZTATB27

The exact model from adb is Lenovo TB-7104F 

Thank you.

erkki-silvola, 2019-06-07:
Try this special build from sdk, where there is check for TB-7104F for scanning

[polar-ble-sdk.aar.zip](https://github.com/polarofficial/polar-ble-sdk/files/3265032/polar-ble-sdk.aar.zip)

SamerDiab, 2019-06-07:
Thank you @erkki-silvola , this is confirmed working.

closed 2019-06-07 by SamerDiab