polarofficial/polar-ble-sdk #24: Does the latest SDK support objective-c?

wujunyang, 2019-06-23:
If there can be a corresponding example

erkki-silvola, 2019-06-24:
In polarBleSdkTestApp there is a class ApiWrapperSwift which shows how to do "bridge" to obj-c.

wujunyang, 2019-06-24:
@erkki-silvola 

I really feel your reply, because I only use OC now. Do you have more detailed information, such as how to use OC to call ApiWrapperSwift ?

erkki-silvola, 2019-06-24:
import generated header #import "ProjectName-Swift.h" ( like 
#import "polarBleSdkTestApp-Swift.h" ) use ApiWrapperSwift like @property ApiWrapperSwift* api;
Notice callbacks does not have a wrapper in testApp

wujunyang, 2019-06-24:
#import "test.h"
#import "polarBleSdkTestApp-Swift.h"

@interface test()

@property(nonatomic,strong)ApiWrapperSwift *api;
@property(nonatomic,strong)PolarBroadcastData *myData;

@end

@implementation test

-(void)loadData
{
    self.api=[[ApiWrapperSwift alloc]init];
    
    [self.api startListenPolarHrBroadcast:^(PolarBroadcastData * _Nonnull data) {
        
    }];
}

@end

------------------------------------------------------
Notice callbacks does not have a wrapper in testApp
------------------------------------------------------

So now I can't get it.（ PolarBroadcastData Hr）？Is there any way to get it?

erkki-silvola, 2019-06-24:
You can get PolarBroadcastData, as that is the obj-c wrapper for PolarHrBroadcastData from swift. What I did mean is that there are no wrappers in the example for PolarBleApiObserver, PolarBleApiPowerStateObserver etc...

wujunyang, 2019-06-24:
@erkki-silvola 

-(void)loadData
{
    self.api=[[ApiWrapperSwift alloc]init];
    
    [self.api startListenPolarHrBroadcast:^(PolarBroadcastData * _Nonnull data) {
       
        data.hr ??
        
    }];
}

Thank you very much for your reply. Now I just need to get the HR value of the device. I try to use the above code, but I can't seem to get the HR attribute. For example, how can I get the HR data, such as the code of the example above?

erkki-silvola, 2019-06-24:
You need to add @objc(like "@objc class PolarBroadcastData : NSObject {
   @objc  let name: String
    @objc let hr: Int" etc... ) to members too, I had forgotten to add those as ApiWrapperSwift class is only a example

wujunyang, 2019-06-24:
@erkki-silvola 

Thank you very much. It's available now. I'll try to connect with the device in the evening.

closed 2019-06-26 by wujunyang