polarofficial/polar-ble-sdk #27: Failed to verify bitcode in PolarBleSdk.framework/PolarBleSdk

Roberdan, 2019-06-30:
hi there, I'm getting this message: "Failed to verify bitcode in PolarBleSdk.framework/PolarBleSdk" and haven't found details on how to handle it, do you have any? thanks,RD

erkki-silvola, 2019-07-01:
Can you give more detail what where you doing when you got that message?

Roberdan, 2019-07-01:
at the end I found the solution removing the i386 with lipo. The problem is that when you submit to apple store it's rejected as the polar code can't be signed. Removing i386 with lipo works but it's quite annoying to do, could you try to solve at the core? thx

erkki-silvola, 2019-07-03:
This is difficult one to solve in the framework, maybe guidance for this is sufficient.

Roberdan, 2019-07-03:
basically you have to go inside the single framework folder and then run:
lipo -info PolarBleSdk (repeat this for each framework from polar)

it will show:" Architectures in the fat file: PolarBleSdk are: i386 x86_64 armv7 arm64"

then u go for:
lipo -remove i386 PolarBleSdk -o PolarBleSdk
lipo -remove x86_64 PolarBleSdk -o PolarBleSdk

do the same for RxAtomic.framework and RxSwift.framework

erkki-silvola, 2019-07-04:
Yes, what I did mean is that I don't want to distribute two separate frameworks one for development and one for the app store distribution. I could add a script and documentation how to add it to application project.

closed 2020-05-09 by Roberdan