polarofficial/polar-ble-sdk #30: In iOS:Always getting NotificationNotEnabled, when call the fetchStoredExerciseList

SumeetMourya, 2019-07-22:
When I connect device with method **connectToDevice** and its delegate method called once device is connected, that this in method  **deviceConnected(_ polarDeviceInfo: PolarDeviceInfo)** I want to create some kind of flow for StopRecording, readList, Remove, StartRecording.

SumeetMourya, 2019-07-22:
For Stop Monitoring PolarH10:

`func stopMonitoringHR() {
        
        self.requestRecordingStatus(connectedDeviceId, success: { (statusValue, entryIDValue) in
            
            //If activity already recording.....
            if (statusValue) {
                
                self.stopRecording(self.connectedDeviceId, success: {

                    self.fetchStoredExerciseList(self.connectedDeviceId, next: { (entry) in
                        self.entry = entry
                        
                        guard let e = self.entry else {
                            return
                        }

                        self.fetchExercise(self.connectedDeviceId, entryValue: e, success: { (value, samples) in
                            self.delegate?.sendExerciseData(hrData: samples)

                            self.removeExercise(self.connectedDeviceId, entryValue: e, success: {
                                
                                self.disconnectFromDevice()
                                self.entry = nil
                                self.activityAlreadyStarted = false

                            }, error: { (errorValue) in
                                NSLog("H10PolarManager:     removeExercise failed: \(errorValue)")
                            })

                        }, error: { (errorValue) in
                            NSLog("H10PolarManager:     fetchExercise failed: \(errorValue)")
                        })
                        
                    }, error: { (errorValue) in
                        NSLog("H10PolarManager:     fetchStoredExerciseList failed: \(errorValue)")
                    })
                    
                }, error: { (errorValue) in
                    NSLog("H10PolarManager:     stop recording failed: \(errorValue)")
                })
            
            //If activity recording not started yet.....
            } else {
                
                    self.fetchStoredExerciseList(self.connectedDeviceId, next: { (entry) in
                        self.entry = entry
                        
                        guard let e = self.entry else {
                            return
                        }
                        
                        self.fetchExercise(self.connectedDeviceId, entryValue: e, success: { (value, samples) in
                            self.delegate?.sendExerciseData(hrData: samples)
                            
                            self.removeExercise(self.connectedDeviceId, entryValue: e, success: {

                                self.disconnectFromDevice()
                                self.entry = nil
                                self.activityAlreadyStarted = false

                            }, error: { (errorValue) in
                                NSLog("H10PolarManager:     removeExercise failed: \(errorValue)")
                            })
                            
                        }, error: { (errorValue) in
                            NSLog("H10PolarManager:     fetchExercise failed: \(errorValue)")
                        })
                        
[... 62 lines cut ...]
                            self.startRecording(self.connectedDeviceId, exerciseId: "TEST_APP_ID", interval: 1, sampleType: 1, success: {
                                NSLog("H10PolarManager:     start successfully")
                            }, error: { (errorValue) in
                                NSLog("H10PolarManager:     start recording failed: \(errorValue)")
                            })
                            
                        }, error: { (errorValue) in
                            NSLog("H10PolarManager:     removeExercise failed: \(errorValue)")
                        })
                        
                    }, error: { (errorValue) in
                        NSLog("H10PolarManager:     fetchStoredExerciseList failed: \(errorValue)")
                    })
                }
                
            }) { (errorValue) in
                NSLog("H10PolarManager:     recording status request failed: \(errorValue)")
            }
            
        } else {
            activityAlreadyStarted = false
            self.connectToDevice()
        }
}
`

erkki-silvola, 2019-07-22:
Have you enabled ftpFeature?

SumeetMourya, 2019-07-22:
@erkki-silvola  Yes, with all the features.
`    var api = PolarBleApiDefaultImpl.polarImplementation(DispatchQueue.main, features: Features.allFeatures.rawValue)
`

erkki-silvola, 2019-07-23:
Do you get ftpFeatureReady callback?

SumeetMourya, 2019-09-08:
@erkki-silvola yes but only when I paired the device. which is asking in: **Bluetooth Pairing Request Dialog.**

I have paired PolarH10 in my application setting. And once I paired the device with help of CoreBluetooth framework.
My issue is I don't have **Bluetooth Pairing Request Dialog.** 
When I start my activity and want to use the PolarH10 functionality like **startRecording** after some time I am getting the **Bluetooth Pairing Request Dialog.**   which is not good

How can I do ask use to **Bluetooth Pairing Request Dialog.**  when I paired my device.

erkki-silvola, 2019-09-09:
Hi, that is iOS side feature to handle attribute writes which require encryption/authentication. This cannot be avoided as there is no API around this.

SumeetMourya, 2019-09-10:
@erkki-silvola I got the point.
I am handling it, but only for the pair button on this pop up 
And is the device is paired then this  **func ftpFeatureReady(_ identifier: String){**
called with update, so I can perform the operation accordingly.

This time my question is:  Can we get any kind of callback/delegate so that I will perform the operation for this situation.

erkki-silvola, 2019-09-11:
So if I understood you would like to have callback just before CCC writes which requires user action ?

SumeetMourya, 2019-09-11:
Yes, you got exactly what I was looking for.

erkki-silvola, 2019-09-11:
I can add this feature for next release, thought cannot yet promise time schedule for this...

SumeetMourya, 2019-09-11:
Thank you, I am glad that you guys are maintaining the code according to other user requirements also.
No issue, right now I am managing my requirement with some other way.
Once it will me done, then I'll incorporate in my application.

closed 2021-04-01 by SumeetMourya