polarofficial/polar-ble-sdk #32: OH1 continues to send HR notifications when charging

vanvalkenburg, 2019-08-14:
When I connect my app (using Polar SDK) to the OH1 I am wearing, it receives HR notifications (`hrNotificationReceived` callback) as expected.

If I then remove the sensor from the armband and click it in the charger cradle, the connection stays alive and my app continues to receive HR notifications, continuously the same as the last correct measurement. And since skin detection is not supported, there is no way to distinguish this from normal measurements.

I observed that the LED is turned off as soon as the sensor starts charging, and accelerometer data stops streaming. HR notifications continue though. HR data notifications only stop when unplugging the sensor from the charger.

erkki-silvola, 2019-08-16:
Hi,

This is a desired functionality in OH1, as mobile can load training data from OH1 even it is in charger.

vanvalkenburg, 2019-08-16:
Is it desirable functionality that the OH1 continues to send notifications for the heart rate (repeating the last measured heart rate), even though the PPG sensor is off?

If so, how do you suggest that we can distinguish these heart rate notifications from valid heart rate values?

erkki-silvola, 2019-08-30:
There could be improvement to make in OH1... For now only way to get information about would be that the SDK would inform that the device is in charging mode, this feature is not yet supported, but I can add it to next release.

erkki-silvola, 2019-11-27:
when we get next OH1 firmware update out , there will be partial fix for this, when charging hr is set to 0

closed 2021-04-25 by JOikarinen