polarofficial/polar-ble-sdk #34: Cannot get PPI that is over 20minutes

rikuriver0, 2019-08-30:
I am trying to get PPI from OH1(firmware2.0.11).But I can get  only 20 minutes PPI.
Is this due to internal logic of OH1? If it is not so, Do you have any idea about this problem?
There are no error in my logcat and other data(HR) continue being got.

erkki-silvola, 2019-08-30:
One reason could be if there is too much movement which should raise PolarOhrPPIData.blockerBit

rikuriver0, 2019-09-02:
Thanks for fast reply.
I get PPI every 5seconds and returned value of blockerBit is always false. but it starts streaming and stops 20minutes later.
If the reason is that,how to get long time data?

erkki-silvola, 2019-09-10:
Hi,

We found out that this is an issue in OH1 firmware, and until this is fixed as a workaround you can try restarting PPI measurement after every 20mins?

rikuriver0, 2019-09-11:
I hava already tried restarting PPI measurement and I succeeded to get PPI more than 20mins.
But there is a problem. it takes 20seconds until restart. The blank data which is caused by restart is important for us.
I hope that firmware issue is fixed in the near future.
Thank you for having you tell me a cause.

closed 2019-09-11 by rikuriver0