polarofficial/polar-ble-sdk #35: ECG Frame Types

dumbshow, 2019-09-01:
Hi,

The data specification shows an ECG frame having 3 bytes.  Does this correspond to one sample, or three samples?  What is the encoding?

https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf

Page 5 indicates that ECG sample "68 00 00" corresponds to 104 microvolts.  How is this encoded?

If this question is better asked in another forum, please let me know.  Thank you.

erkki-silvola, 2019-09-02:
Hi,

Data is transferred in big endian so "68 00 00" = 0x000068 signed 24bit int. In the document data is in base 16 = hex.

dumbshow, 2019-09-02:
Thank you.  Shouldn't the final line of this python code be equal to 104 microvolts, as in the documentation?  Am I misunderstanding something?

In [150]: bytes([int(ii, 16) for ii in '68 00 00'.split()])
Out[150]: b'h\x00\x00'

In [151]: int.from_bytes( b'h\x00\x00', byteorder='big', signed=True)
Out[151]: 6815744

My apologies for the naive questions.

closed 2019-09-03 by dumbshow