polarofficial/polar-ble-sdk #38: PPI timestamp property always reported as 0. (Polar iOS Demo app with OH1)

marchv, 2019-09-26:
Hi Polar team :)

I have just updated my OH1 to latest firmware and downloaded your iOS-SDK. I then wanted to get PPI samples and added some prints in the ppiToggle(_ sender:) method:

```
                case .next(let data):
                    print("data timestamp: \(data.timeStamp)") // always zero!
                    for item in data.samples {
                        NSLog("PPI: \(item.ppInMs); errorEstimate: \(item.ppErrorEstimate)")
                    }
```

But 'date.timestamp' is always zero!?

Or have I misunderstood the interface?

I also tried to call setTime (succeeded) once connected but that made no difference...

The reason for this is that I wanted to try to export the Beat-To-Beat-Interval to Apple's HealthKit (iOS 13). But maybe this is not possible...

BTW: I assume the errorEstimate is in milliseconds?

Thanks!

erkki-silvola, 2019-09-30:
Hi,

We found issue in OH1 firmware regarding ppi timestamps...

FloNOC, 2019-11-05:
Hi,

The whole documentation seems to indicate that the timestamp for OH1 PPI will always be 0.

So is it a bug or a feature ? ;-)

erkki-silvola, 2019-11-05:
It is a bug in OH1, and it seems that it cannot be even fixed

FloNOC, 2019-11-05:
Oh, ok.

I thought it was linked with the fact that the OH1 seems not to have a "clock", as there is no way to set it throught BLE, is it ?

erkki-silvola, 2019-11-07:
There is epoch on the OH1, it is just an internal software structure which affects that there is no way to add accurate timestamp for ppi

FloNOC, 2019-11-07:
Oh, OK.

Thanks for the info, we'll keep on using the reception device's timestamp then.

marchv, 2019-11-07:
@erkki-silvola thanks for being open about the problem in OH1 👍

closed 2021-04-25 by JOikarinen