polarofficial/polar-ble-sdk #39: Not receiving battery charge notifications when connecting via Polar SDK to OH1

vanvalkenburg, 2019-09-30:
When I connect to the OH1 from my Android app using the standard Battery Profile, I see notifications coming in for changes in the battery charge percentage (in multiples of 10), as expected.

However, if I use the Polar SDK, I get a callback to `batteryLevelReceived` upon connection, but never any update to the battery charge after that. HR and accelerometer data are received as expected.

erkki-silvola, 2019-10-02:
current implementation is intended to work like that there is only once invoked callback of the battery level after connection, but I can change the implementation to invoke updates too as I can understand that there can be use cases for it...

bhatramya, 2019-11-07:
Dear Polar team,

Is the implementation already changed to receive the battery level updates regularly? Please let me know when it is changed.

Thank you.

erkki-silvola, 2019-11-08:
Hi,

It should work with the latest release

bhatramya, 2019-11-08:
Thank you. It works! 👍

vanvalkenburg, 2019-11-27:
This is working well now, thanks!

vanvalkenburg, 2020-06-19:
There seems to be a regression on this as I can reproduce it on the latest version (v2.2.2).

Now that the code has been published (👍), I can see that the battery characteristic notifications are handled through a `Single` instead of `Flowable`.

marmarelis, 2021-02-21:
Hello, I would be grateful for a fix to this as well.

closed 2021-02-23 by JOikarinen

JOikarinen, 2021-02-23:
Bug shall be now fixed in [71cf6e8](https://github.com/polarofficial/polar-ble-sdk/commit/71cf6e85c65bea78968b55de1a672af8e6b56fcc). Originally fixed in 2.2.1 but was unfortunate reintroduced in release 2.2.2.

marmarelis, 2021-02-23:
Thanks. From what I can tell, you updated the Android framework. If I am not mistaken, the bug still persists in the iOS version.

JOikarinen, 2021-02-24:
@marmarelis filed new [issue](https://github.com/polarofficial/polar-ble-sdk/issues/130) to resolve the iOS-SDK too.

JOikarinen, 2021-03-02:
Previous fix [71cf6e8](https://github.com/polarofficial/polar-ble-sdk/commit/71cf6e85c65bea78968b55de1a672af8e6b56fcc) introduced another regression bug. As BLE Battery Service client may get the battery status from BLE Battery Service before the any SDK client has started observing the changes  it is good to cache any received battery status. Fixed in [1ea946c](https://github.com/polarofficial/polar-ble-sdk/commit/1ea946c7297ca98579d9529b4910a5149c79117d)

commit 71cf6e85 (2021-02-23): Fix #39: Battery status is now continuously published