polarofficial/polar-ble-sdk #40: Receiving ECG microvolts more quickly

dumbshow, 2019-10-20:
Hi,

I am using this command (after preliminaries) to request an ECG stream from a Polar H10.

`char-write-req 0x0047 02000001820001010E00`

The data is returned in a format like this, which seems to contain 73 samples representing microvolts: 00 a0 7d c8 a5 8c 44 52 08 .....

My problem is that I'm trying to detect the onset of heartbeats in real time.  By the time this data is received, the heartbeat has passed.  Is it possible to receive this data more quickly, perhaps in smaller batches?  I have tried changing the "resolution" and "sample rate" parameter but have not been able to make it work:

`char-write-req 0x0047 02000001820001010C00`

Thank you for your help.

erkki-silvola, 2019-10-25:
Hi,

I understand your case. Minimum mtu for H10 should be 64, but I will confirm this on monday. Reason we packed as much as possible to get maximum through put... especially if we central side supports packet length extension...

dumbshow, 2019-10-25:
Thank you, Erkki.

NiwreRelbag, 2019-11-04:
Hi,

we are also interested in a "close to real time" analysis of the ECG data. For an MTU of 64 bytes: How many ECG samples will then fit into a single package? Do you plan to extend the SDK for a configuration of the block size?

Thanks for help, Niwre

erkki-silvola, 2019-11-04:
Hi,

That can be done but it would be only possible on android because on iOS os controls the mtu exchange.

NiwreRelbag, 2019-11-05:
Oh, that's bad news - our app currently is iOS only :-).
Thanks for the response.

mul-12, 2020-06-22:
Hi, Just wondering if you could provide a little support regarding the preliminaries mentioned in the first post (so that I can start receiving ECG data). When entering "char-write-req 0x0047 02000001820001010E00" the characteristic is written successfully, however I'm not receiving any ECG data. I have looked at the data spec, but could use a hand. Thanks!

closed 2022-02-11 by JOikarinen