polarofficial/polar-ble-sdk #43: Cannot get PPI for longer than 20 minutes

FloNOC, 2019-11-06:
Hello,

It is the same issue as exposed here : https://github.com/polarofficial/polar-ble-sdk/issues/34
This issue is closed, but the problem still exists.

We have made a workaround in our application, but it results in having 20s gaps in measurements, which is very annoying...

As much as I know that it is a firmware issue, is it possible to use this one as a follow-up for when it's fixed ?
If not, just close it ;-)

Thank you

erkki-silvola, 2019-11-07:
There will be a fix for this but I cannot promise any time schedule for this yet...

pro-g111, 2019-11-07:
I am having the same issue as @FloNOC.

FloNOC, 2019-11-07:
> There will be a fix for this but I cannot promise any time schedule for this yet...

Again, thank you for your work and the infos. In case this bug get fix in any way, please please update this issue, which I'll keep open for that purpose.

This one is pretty critical for us to use polar devices for our application.

AmbroiseRecht, 2020-01-22:
We have the same issue and we are very interesting in the fix.

Any update about this ticket ?
Do you have news about a schedule for the fix ?

erkki-silvola, 2020-01-22:
Hi, 

The fix is coming but no exact schedule yet available, I will inform later....

erikboto, 2020-04-16:
Any news, good or bad?

erkki-silvola, 2020-04-16:
Hi,

Fix has been already made but there is no schedule yet when release will be rolled out...

AmbroiseRecht, 2020-05-13:
Hi,

Thank you for the information @erkki-silvola 
Do you have more information about the release date of this new firmware ?
Do you know if we will have just to update the firmware or if we will have to change the SDK also ?

JOikarinen, 2021-04-25:
The [firmware 2.1.20](https://support.polar.com/en/updates/polar-oh1-firrmware-update-2120?product=OH1) to OH1 shall fix the problem

closed 2021-04-25 by JOikarinen