polarofficial/polar-ble-sdk #44: cannot find symbol method autoConnectToPolarDevice

astralmedia, 2019-11-13:
In my java class, I declare the variable:

`PolarBleApi api = PolarBleApiDefaultImpl.defaultImplementation(reactContext, PolarBleApi.FEATURE_HR);`

then in my method I hit later, I call autoConnect:

`api.autoConnectToPolarDevice(-50, null).subscribe();`

I just keep getting this error:

```PolarModule.java:70: error: cannot find symbol
        api.autoConnectToPolarDevice(-50, null).subscribe();
           ^
  symbol:   method autoConnectToPolarDevice(int,<null>)
  location: variable api of type PolarBleApi
1 error```

Thanks for any insight.

erkki-silvola, 2019-11-14:
Hi,

It has been renamed a while ago to autoConnectToDevice

astralmedia, 2019-11-14:
Thank you! FYI the Readme main android example has the old method.

mohataher, 2021-03-23:
Please reconsider updating the README once again. The default signature is deprecated.

```
polarBleApi.autoConnectToDevice(-50, null); // cannot resolve method 
```

JOikarinen, 2021-03-29:
@mohataher fixed in https://github.com/polarofficial/polar-ble-sdk/commit/40ae0c4d934a117b67f66e520510dbdea4e2fbd6.

closed 2021-03-29 by JOikarinen