polarofficial/polar-ble-sdk #46: PC-Support

fre4kyC0de, 2019-11-23:
Is there any plan to add support for computers (Windows, Linux, ...)?

Currently I‘m using my own implementation to interpret the data using Bluetooth GATT (HeartRate). It would be great to get an official SDK, which would allow to view even more data (such as the ECG signal).

jussivirkkala, 2019-11-23:
You can use https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf for additional data with own implementation

willianpaixao, 2020-01-22:
Agreed, official SDK support for PC is a must feature.

pablomainar, 2020-11-19:
Any news on this?

JOikarinen, 2021-04-25:
At the moment this is our official statement: "Unfortunately, we are only able to support Android and iOS. The SDK includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for."

closed 2021-04-25 by JOikarinen