polarofficial/polar-ble-sdk #47: Unable to start PPG streaming on OH1

anlaital, 2019-11-30:
Running the example app (which by the way crashes on launch as it does not contain the correct privacy key for Bluetooth in its Info.plist) and connecting to an OH1 with it and tapping "toggle ppg" always fails with the following error:

`polarBleSdkTestApp[5289:2696653] start ppg error: PolarBleSdk.NotificationNotEnabled`

The same thing happens when you try to use the SDK to start PPG streaming as it always fails when invoking `api.requestPpgSettings`.

There is also no documentation that I could find regarding the possible error codes returned by the SDK. It would be highly useful to document these.

erkki-silvola, 2019-12-02:
Hi, I updated the iOS example app's info plist. I had missed iOS 13+ changes regarding bt (example app is just show as simple as possible how to use the SDK). There is documentation in the docs folder, also Xcode shows docs in the quick help too.

erkki-silvola, 2019-12-02:
What is the OH1 firmware in your device?

anlaital, 2019-12-02:
I managed to get this working by buying a newer hardware version of the OH1. The previous device was could not be discovered with Polar Flow and it had an old firmware version.

With the new OH1 and firmware version 2.0.10 everything seems to work okay.

closed 2019-12-02 by anlaital