polarofficial/polar-ble-sdk #48: Unable to stream ECG data?

DataSupply, 2019-12-04:
I've downloaded the SDK and installed the iOS example.
I can succesfully 'manually' connect the H10 sensor with the app and retrieve HR readings after changing the deviceId.
However when pressing the "Toggle ECG" i always get the error: 

**start ecg error: PolarBleSdk.NotificationNotEnabled**

By default the api is initialized with the option "Features.allFeatures.rawValue" (default) as well as the "Features.polarSensorStreaming.rawValue" setting. 

From my understanding the ECG feature first needs to requested. Which is done by the pressing the toggle ecg button..

Not sure if it's relevant but after connecting these services logged 
2019-12-04 14:46:02.920771+0100 polarBleSdkTestApp[1301:331205] [BLE] service discovered:  Heart Rate
2019-12-04 14:46:02.920968+0100 polarBleSdkTestApp[1301:331205] [BLE] service discovered:  181C
2019-12-04 14:46:02.921222+0100 polarBleSdkTestApp[1301:331205] [BLE] service discovered:  Device Information
2019-12-04 14:46:02.921448+0100 polarBleSdkTestApp[1301:331205] [BLE] service discovered:  FEA5
2019-12-04 14:46:02.921852+0100 polarBleSdkTestApp[1301:331205] [BLE] service discovered:  Battery
2019-12-04 14:46:02.922402+0100 polarBleSdkTestApp[1301:331205] [BLE] service discovered:  FEEE

What am i missing to get the ECG stream working?

erkki-silvola, 2019-12-09:
Hi,

What is the firmware version in your H10 ?

DataSupply, 2019-12-10:
hi Erkki-silvola, Thank you for your comment. You triggered me to double check the firmware, and after googling it turned out that indeed still had an old firmware version. Important to note is that the Polar beats app didn't show an update popup until i logged in with an account. So it seems that no firmware upgrade can be installed by using the app as a guest user.. After the firmware upgrade everything worked as expected 👍

closed 2023-09-08 by jimmyzumthurm