polarofficial/polar-ble-sdk #51: Auto receiving HR data

astralmedia, 2019-12-18:
I am noticing that the HR hrNotificationReceived callback is automatically receiving data once I connect to a device. Is there a way to disable that and only run when I activate it?

Also, I am going to be connecting to different types of Polar devices...is the HR signal the common thread between them all? Or do I need to specifically sync up to their PPG/ECG streams based on device?

Thanks so much

erkki-silvola, 2019-12-30:
Hi,

There is no API for stop/start hr receiving, only the feature bit which "totally" disable or enable it.
HR is common at least on the SDK scope. Can you give more detail what you mean by "Or do I need to specifically sync up to their PPG/ECG streams based on device?"

astralmedia, 2020-01-07:
I had misunderstood how the PPG/ECG relates to HR. I have figured out how to get the data coming in :)

closed 2020-01-07 by astralmedia