polarofficial/polar-ble-sdk #53: Data seems to be coming at low hz from OH1 & H10

astralmedia, 2020-01-07:
I am getting the data coming back properly from both OH1 and H10, but it seems more like 15-30hz, not the 130hz I've read they can do.

I am starting both PPG and ECG signals with the max settings, but still the data comes slower than expected.

Am I just missing some settings somewhere?

erkki-silvola, 2020-01-09:
How are you calculating the sample rate?
Here is a simple example in java how to calculate sample rate notice this is more of "pseudo" code:

`api.startAccStreaming(identifier, settings).buffer(2, 1).map(new Function<List<PolarAccelerometerData>, List<Pair<Long, PolarAccelerometerData.PolarAccelerometerSample>>>() {
            @Override
            public List<Pair<Long, PolarAccelerometerData.PolarAccelerometerSample>> apply(List<PolarAccelerometerData> datas) throws Exception {
                long delta1 = ((datas.get(1).timeStamp - datas.get(0).timeStamp) / datas.get(1).samples.size());
                long baseTimeStamp1 = datas.get(0).timeStamp - ((datas.get(0).samples.size()-1)*delta1);
                Log.d(TAG,"aac: " + (1.0f/(((float)delta1)/1000000000.0f)));
                List<Pair<Long, PolarAccelerometerData.PolarAccelerometerSample>> list = new ArrayList<>();
                for(PolarAccelerometerData.PolarAccelerometerSample sample : datas.get(0).samples){
                    list.add(new Pair<>(baseTimeStamp1,sample));
                    baseTimeStamp1 += delta1;
                }
                return list;
            }
        });`

astralmedia, 2020-01-09:
I am calculating the delta between signals by keeping track of the length of time between signals coming in and tracing it out each time a signal comes in.

I am basically using the sample code from the github android example adjsuted slightly for our code:

`ppgDisposable = api.requestPpgSettings(DEVICE_ID).toFlowable().flatMap(new Function<PolarSensorSetting, Publisher<PolarOhrPPGData>>() {
                @Override
                public Publisher<PolarOhrPPGData> apply(PolarSensorSetting polarPPGSettings) throws Exception {
                    Log.d("ReactNative","Start PPG Stream");
                    return api.startOhrPPGStreaming(DEVICE_ID, polarPPGSettings.maxSettings());
                }
            }).subscribe(
                new Consumer<PolarOhrPPGData>() {
                    @Override
                    public void accept(PolarOhrPPGData polarOhrPPGData) throws Exception {
                        for(int i = 0; i < polarOhrPPGData.samples.size(); i++) {
                            //Log.d("ReactNative","PPG : " + polarOhrPPGData.samples.get(i).ppg0);
                            if(i == (polarOhrPPGData.samples.size()-1)){
                                PolarOhrPPGData.PolarOhrPPGSample data = polarOhrPPGData.samples.get(i);
                                //Log.d("ReactNative", "ppg0: " + data.ppg0 + " ppg1: " + data.ppg1 + " ppg2: " + data.ppg2 + "ambient: " + data.ambient);
    
                                if(time == null) {
                                    time = Instant.now(); // for next one
                                } else {

                                    Instant now = Instant.now();
                                    long delta = Duration.between(time, now).toMillis();
                                    time = Instant.now(); // for next one

                                    Log.d("ReactNative", "Signal DELTA " + delta);
                                }                                
                                // Create ArrayList<Int> to send back
                                ArrayList<Integer> signalValues = new ArrayList<>();
                                signalValues.add(data.ppg0);
                                signalValues.add(data.ppg1);
                                signalValues.add(data.ppg2);
                                signalValues.add(data.ambient);
                                ipcEventSubsystem.signalUpdateIntArray(getDeviceEventName(), "ppg", signalValues);
                                //Log.d("ReactNative", getDeviceEventName());
                            }
                        }

                        
                    }
                },
                new Consumer<Throwable>() {
                    @Override
                    public void accept(Throwable throwable) throws Exception {
                        //Log.e("ReactNative","PPG#: "+throwable.getLocalizedMessage());
                        Log.e("ReactNative","PPG#: "+throwable.toString());
                    }
                },
                new Action() {
                    @Override
                    public void run() throws Exception {
                        Log.d("ReactNative","complete");
                    }
                }
            );`

Also, using `api.startOhrPPGStreaming(DEVICE_ID, polarPPGSettings.maxSettings());`

I am sure its not 130hz or even close.

astralmedia, 2020-01-10:
Can anyone help me understand how to dive into the PolarSensorSetting parameters? Not sure how to dive into the Map to understand what the settings are coming back as.

erkki-silvola, 2020-01-13:
Hi, PolarSensorSetting contains settings which are available in the the device, where key is SAMPLE_RATE, RESOLUTION or RANGE and value is Set containing all possible values supported by the key....

astralmedia, 2020-01-14:
Thanks for the input, I am going to do some more testing shortly and get back here...I am not sure I have the full hz coming in but have an idea.

closed 2020-01-29 by astralmedia

arianesasso, 2020-11-20:
Hi, sorry for posting on a closed issue, but could you let me know what is the sampling frequency from the PPG channels in Polar OH1? Thank you.