polarofficial/polar-ble-sdk #55: H10 ecg signal update speed

astralmedia, 2020-01-29:
After quite a bit of experimenting, we have narrowed down some weirdness we are experiencing. Basically the sensor itself is working at ~130hz, but the updates coming back from the API seem to be chunked together in groups, coming at a much lower hz (around 2-3)...the resolution of the data is very clean, however for realtime application of this, it is much too low to be usable.

https://www.youtube.com/watch?v=EaO-Qd216p8 This video goes over the emerging medical & biofeedback research market, and the reasons for the need of realtime sensors.

I am wondering if there is any way to adjust the api to send smaller, more frequent updates from the H10. The OH1 is coming in smaller groups of data much faster (25+hz at its native data resolution) and is working great...however the ECG is a necessary part of our workflow.

Thanks for any insight, and again, thank you for such a great line of products!

erkki-silvola, 2020-02-06:
Hi,

Only way to lower latency is to change mtu, but this cannot be done in iOS but in Android can...

gavaga, 2022-02-10:
I know this is old, but if anyone else comes across this, I've been working with this lately, and on the Polar H10, I found the minimum MTU allowed by the device is 70 bytes (3 BLE, 10 Header, 57 Data = 19 samples), which gives me around 7-8Hz updates from the device.

JOikarinen, 2022-02-11:
@gavaga thanks for the update. I had similar sum up in another issue, I will add it here.

From https://github.com/polarofficial/polar-ble-sdk/issues/227#issuecomment-1020961151:

With 130Hz sampling each sample is spaced ~7.69ms apart. The number of samples in one ECG data package depends on what kind of Bluetooth connection parameters are negotiated i.e. what is the Bluetooth data packet size used on BLE communication channel (in BLE terminology the data packet size on communication is called MTU, maximum data unit). The default MTU with H10 is 232 bytes and it results each ECG package contains 73 ECG samples. Roughly it means that one ECG package contains half second of sampled data (73samples/130Hz = 0.56s). If you are using Android as you environment you have option to reduce the MTU size with API function [setMtu()](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L131). If you set the MTU to minimum 70, then each ECG data packet size contains 19 samples. 19samples/130Hz = 0.146s. Please note, that MTU size can be set only on Android and setMtu() has to be called before connection is created.

closed 2022-02-11 by JOikarinen