polarofficial/polar-ble-sdk #56: SDK Library Bindings for Xamarin Android and iOS

tessarolli, 2020-02-13:
Hi Polar Developers,
Have any of you been able to create necessary bindings for the native libs on Android and iOS projects in a Xamarin / Xamarin Forms solution???

I've been trying to integrate Polar Sensors Data into my xamarin forms application and i wonder if someone already done this before?

tessarolli, 2020-02-17:
OK, i've done it.

For future reader reference, here's how you do it:
Add a new Android Binding project to solution.
Add the polar-ble-sdk.aar to JARS folder, and set build action to LibraryProjectZip
Add a new class file named AttributeOperation.cs to additions folder and set this to its content:
using Java.Lang;
namespace Com.Androidcommunications.Polar.Enpoints.Ble.Common.Attribute
{
public partial class AttributeOperation : Object, IComparable
{
int IComparable.CompareTo(Object obj)
{
return CompareTo((AttributeOperation)obj);
}
}
}

Add two nugget references: a) Naxam.RxAndroid.Droid - b) Naxam.RxJava2.Droid

Build the project, and reference this project on your Xamarin.Android project.

And thats it for mapping the native android library to you xamarin android project.

Now for some code samples on how to interface with a polar sensor, this is what i've used:
(Please, have in mind that this my first approach on this subject, and this code is only in early stage of development, thus its only saving a string representation of the data received on a List of strings)

`using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;

using Android.App;
using Android.Content;
using Android.OS;
using Android.Runtime;
using Android.Views;
using Android.Widget;
using Polar.Com.Sdk.Api;
using Polar.Com.Sdk.Api.Model;

namespace TessarolliSolutions.Droid.Services.PolarSDK
{
    public class PolarSDK
    {
        PolarBleApi api;
        List<string> Log;

        public PolarSDK()
        {
            Log = new List<string>();
            // Verificar Quais Features vai utilizar ao instanciar a api
            api = PolarBleApiDefaultImpl.DefaultImplementation(MainActivity.Instance, PolarBleApi.FeatureBatteryInfo | PolarBleApi.FeatureDeviceInfo | PolarBleApi.FeatureHr);

            api.SetApiCallback(new PolarBleApiCallbackImpl(Log));

            var autoconnect = api.AutoConnectToDevice(-50, "180D", null).Subscribe(new RxAction(() => { Log.Add("Concluido"); }), new RxConsumer<Java.Lang.Throwable>((e) => { Log.Add($"Error: {e}"); }));

            //var scan = api.SearchForDevice(new Consumer(Log));
            
        }

        public class RxAction : Java.Lang.Object, IO.Reactivex.Functions.IAction
        {
            private readonly Action Callback;
            public RxAction(Action callback)
            {
                Callback = callback;
            }

[... 123 lines cut ...]
        }

        public override void PpiFeatureReady(string identifier)
        {
            base.PpiFeatureReady(identifier);
            Log.Add($"PPI Feature Ready: {identifier}");
        }

        #endregion

        #region Dispose

        protected override void Dispose(bool disposing)
        {
            base.Dispose(disposing);
            Log = null;
        }

        #endregion

    }

}`

I have not tampered with the IOS mapping yet.

erkki-silvola, 2020-02-17:
Hi,

Nice job.

tessarolli, 2020-02-17:
Thanks, there's still a long way to go, but this should be a great starting point for future developers out there.

RicardoMDI, 2020-08-06:
Thanks for the contribution. Can you share your example code? I'm trying to do the same thins but it seems not to work. (I'm Brazilian too, if that matters :) )

RicardoMDI, 2020-08-06:
Add a new class file named AttributeOperation.cs to additions folder and set this to its content:
using Java.Lang;
namespace Com.Androidcommunications.Polar.Enpoints.Ble.Common.Attribute
{
public partial class AttributeOperation : Object, IComparable
{
int IComparable.CompareTo(Object obj)
{
return CompareTo((AttributeOperation)obj);
}
}
}

I'm having trouble here, it says "The name 'CompareTo' does not exist in the actual context". What am I doing wrong?

SuperCorks, 2021-09-22:
Hi @tessarolli, @RicardoMDI , thank you for this thread! It has helped us a lot. Have you had any issues with the `setLocalTime()` method? Or anything [related to this question](https://github.com/polarofficial/polar-ble-sdk/issues/199)? 

Any help would be appreciated!

tessarolli, 2021-09-22:
Hi @SuperCorks , i'm glad it helped!

To be honest, i've never had the chance to finish the project I was working on this, due to a change of priorities.

Maybe I'll revisit it later.

When I was testing it, i've not had any problems with setLocalTime()...

tessarolli, 2021-09-22:
@RicardoMDI sorry about the long time.. maybe it was missing some using directives.. i dont remember it anymore.. sorry again..

SuperCorks, 2021-09-22:
Thank you for the feedback, we're going to open-source the Android, iOS and portable project as soon as we have a working prototype (which we already have for Android except for the `setLocalTime()` method).

SuperCorks, 2021-09-23:
Solved my issue. https://github.com/polarofficial/polar-ble-sdk/issues/199#issuecomment-925883492

closed 2021-09-25 by JOikarinen