polarofficial/polar-ble-sdk #58: "dyld: Symbol not found" with Carthage compiled RxSwift 5

aegis965, 2020-02-24:
I'm on Polar Ble Sak v2.2.1-patch.1, using Carthage compiled RxSwift 5.0.1.
Running on iPhone and simulator gave me this error:
`dyld: Symbol not found: _$s7RxSwift10ObservableCMo
  Referenced from: [...]/Frameworks/PolarBleSdk.framework/PolarBleSdk
  Expected in: [...]/Frameworks/RxSwift.framework/RxSwift
 in [...]/Frameworks/PolarBleSdk.framework/PolarBleSdk`

erkki-silvola, 2020-02-27:
Hi

I need a bit more insight of the setup...

aegis965, 2020-03-02:
Hi,
I'm getting this error under Xcode 11.3.1, 11.2.1, and 11.2, rxswift compiled with Swift 5.1.2. OS version is Catalina 10.15.3.

erkki-silvola, 2020-03-02:
Hi,

Do you have framework search path set correctly?

aegis965, 2020-03-03:
I believe I have.
I also tried using your pre-compiled RxSwift. The app compiles without problem, but when other dependencies like RxCocoa is called, the app sigabrts.

erkki-silvola, 2020-03-19:
Hi, I added the source code dump under sources folder, there is actually also latest compiled RxSwift, but if there is some problems still add the whole project....

aegis965, 2020-03-26:
Hi, I ran `carthage update` in the sources folder and rebuilt the sdk. Now every works fine. Thank you.

manu-alde, 2021-02-11:
Hello! I'm getting a similar error but using cocoapods. I'm using XCode 12.4 with required RxSwift version 6, swift 5.

What I tried is using cocoapods directly on the iOS example you provide for the sdk after a few tries, but it ended with a "RxSwift not module not found" and "Failed to build module 'PolarBleSdk'" from the XCFramework itself.

It might not be a problem of the sdk, but what I would like to know if there is something I'm missing. And also, what would be the best to do? Tried also replacing cocoa's RxSwift with XCFramework one but got into dependency problems as well.

JOikarinen, 2021-02-12:
@manu-alde I was able to reproduce your problem with [iOS example project](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios). With below steps I got the project working. 

1. Open the example project and remove the RxSwift dependency from the project. Closed the project.

2. Added Podfile:
```
# Podfile
use_frameworks!

target 'polarBleSdkTestApp' do
    pod 'RxSwift', '6.0.0'
    pod 'RxCocoa', '6.0.0'
end
```
3. Run the ```pod install```.  The pod install gave the warnings like: 
    ```[!] The `polarBleSdkTestApp [Debug]` target overrides the `FRAMEWORK_SEARCH_PATHS` ``` etc. 

4. To resolve the warnings  open the polarBleSdkTestApp workspace with Xcode and  add  in target  polarBleSdkTestApp -> Build Settings -> added `$(inherited)` for "Framework Search Paths" and Header Search Paths"

**At this point the project builds.  However, if trying to run the app there is crash: "dyld: Symbol not found:..."**

5.  Insert this to end of Podfile and run ```pod update```
```
post_install do |installer|
installer.pods_project.targets.each do |target|
target.build_configurations.each do |config|
config.build_settings['BUILD_LIBRARY_FOR_DISTRIBUTION'] = 'YES'
end
end
end
```

**Now example project builds and runs**

Used these references while investigated the problem:
- https://stackoverflow.com/questions/40599454/use-the-inherited-flag-or-remove-the-build-settings-from-the-target-c
- https://github.com/CocoaPods/CocoaPods/issues/9775#issuecomment-722298424

manu-alde, 2021-02-12:
It worked! I missed the $(inherited) warning when trying this last 'post_install' solution, that did the trick with the last dependency issues. Thank you!

closed 2021-02-13 by JOikarinen