polarofficial/polar-ble-sdk #61: Multiple Sensor HR data response method update request.

jams032, 2020-02-27:
Hello Polar Team,

I  am implementing Polar SDK (iOS) for polar H10 sensor in our existing iOS fitness app. I can perform a session by connecting multiple H10 sensor. Sensor Connection and data providing works accordingly. Data receive are performed by using PolarBleApiDeviceHrObserver and

`func hrValueReceived(_ identifier: String, data: PolarHrData) { } `

method successfully.

Question 1 :
Here, hrValueReceived() method is called in every second for every device one time. But receiving data for n sensor , called the method n times in 1 second which seems increases the complexity for post processing of the data. It's better for SDK to collect all connected sensors data in an array for a seconds and send back to implementation class in every second rather than n times ( in 1 second) .

Does SDK provide this facility right now? How we can make this smooth ? 

Question 2 : Sensor disconnection works fine but if there is a invalid device ID is send to try api.disconnectFromDevice("\(device.DeviceID!)") , it shows a log from SDK. But no delegate method is seen. 

Is it available ? 

Thanks.

erkki-silvola, 2020-02-27:
Hi,

Question 1 :
SDK is only a "dummy" data transfer between sensor and app, and because of how ble works collecting data from multiple sensor and sending the collected chunk is not straight forward to do. So I believe you need to somehow handle this in the app itself.

Question 2 :
Yes PolarBleSdk.InvalidArgument is thrown if deviceId or address is incorrectly formatted and therefor there is no delegate callback invoked.

jams032, 2020-02-28:
Thanks for your reply. Another related question.

We are trying to implement polar SDK in our existing fitness app. 
Polar SDK can connect any HR sensor like Wahoo TICKR because both (Polar H10, Wahoo TICKR) are manufactured for same purposes and perform with same API. Do you have any future plan to stop support connecting other Heart rate sensor (Wahoo TICKR or other ) from Polar SDK ? If not, we could continue with Polar SDK to connect all types of devices.

Thanks.

erkki-silvola, 2020-03-02:
Hi,

The supported features won't be removed once they have been added.

closed 2020-03-03 by jams032

jams032, 2020-05-12:
@erkki-silvola  Hi ,

Polar SDK show pairing Request alert while connecting H10 sensor after calling api.connectToDevice(ID). 

This is annoying for the app we are working with. How to avoid this ?

jams032, 2020-05-19:
Is it related to this ? 

https://stackoverflow.com/questions/29026377/ios-disable-bluetooth-pairing-request-popup