polarofficial/polar-ble-sdk #62: `Features` enum as `OptionSet`?

theoriginalbit, 2020-03-03:
Taking a look at some of your examples ([like this one](https://github.com/polarofficial/polar-ble-sdk/blob/master/demos/iOS-Demos/PolarSDK-Exercise-Demo/PolarSDK-Exercise-Demo/ViewController.swift#L36)) and I'm seeing API features handled like so:

```swift
Features.batteryStatus.rawValue | Features.deviceInfo.rawValue | Features.polarFileTransfer.rawValue
```

It seems to me like this could be instead handled through an [`OptionSet`](https://nshipster.com/optionset/).

It would simplify the contract greatly as the `PolarBleApiDefaultImpl.polarImplementation(_:features:)` function could have a compile-checked signature:
```swift
public static func polarImplementation(_ queue: DispatchQueue, features: Features) -> PolarBleSdk.PolarBleApi {
  // …
}
```
And the caller wouldn't require performing `|` or  `rawValue`:
```swift
/// Many features
PolarBleApiDefaultImpl.polarImplementation(.main, features: [.batteryStatus, .deviceInfo, .polarFileTransfer])

/// Single feature
PolarBleApiDefaultImpl.polarImplementation(.main, features: .allFeatures)
```

Looking at the current enum the `OptionSet` equivalent would be
```swift
struct Features: OptionSet {
    var rawValue: Int

    static let hr                   = Features(rawValue: 1 << 0)
    static let deviceInfo           = Features(rawValue: 1 << 1)
    static let batteryStatus        = Features(rawValue: 1 << 2)
    static let polarSensorStreaming = Features(rawValue: 1 << 3)
    static let polarFileTransfer    = Features(rawValue: 1 << 4)
    static let allFeatures          = Features(rawValue: 1 << 5)
}
```

Though would probably be better implemented as
```swift
struct Features: OptionSet {
    var rawValue: Int

    static let hr                   = Features(rawValue: 1 << 0)
    static let deviceInfo           = Features(rawValue: 1 << 1)
    static let batteryStatus        = Features(rawValue: 1 << 2)
    static let polarSensorStreaming = Features(rawValue: 1 << 3)
    static let polarFileTransfer    = Features(rawValue: 1 << 4)
    
    static let all: Features = [.hr, .deviceInfo, .batteryStatus, .polarSensorStreaming, .polarFileTransfer]
}
```

erkki-silvola, 2020-03-08:
Hi,

Yep, I would do it now that way, but when I originally wrote it it was almost my first swift code. This only a cosmetic thing and the SDK is more or less in maintenance mode, so this won't be changed for now.

theoriginalbit, 2020-03-09:
Fair enough. It would be nice to see this, and a couple other features we've found useful, be integrated back into the SDK. Are there any plans towards open-sourcing?

erkki-silvola, 2020-03-09:
Open sourcing is the next step...

erkki-silvola, 2020-03-19:
source dump has been made you can find under sources folder

theoriginalbit, 2020-03-19:
Awesome, thanks heaps!

theoriginalbit, 2020-03-19:
I presume since the SDK is in maintenance mode you won't be accepting any Pull Requests?

erkki-silvola, 2020-03-20:
Probably not, and bare in mind that the code bases are very old, thus would/will need a lot of rewriting etc...

erkki-silvola, 2020-03-20:
Note also latest RxSwift is recompiled with "libraries for distribution"

theoriginalbit, 2020-03-20:
>thus would/will need a lot of rewriting etc...

Possibly, but there's definitely some low hanging fruit. One of the smallest being this issue, slightly more work (but also not too difficult) Carthage/Cocoapods/SPM distribution. From what I can see only two things which prevents that:
1. RxSwift as a dependency instead of a built framework – super easy
1. ProtoBuf is currently ObjC and for SPM would need to be Swift, which [Apple has a solution for](https://github.com/apple/swift-protobuf)) – needs the original `.proto` to generate

>Note also latest RxSwift is recompiled with "libraries for distribution"

I did notice that

erkki-silvola, 2020-03-20:
**1.  RxSwift as a dependency instead of a built framework – super easy to address**
- true, and at some point combine native library will replace the RxSwift totally

**ProtoBuf is currently ObjC and for SPM would need to be Swift, which Apple has a solution for) – needs the original .proto to generate**
- When I originally compiled protos to iOS only ObjC port was available(long time ago), thought protos are only needed for to read stored execises just saying if you don't need this feature just remove those.... I will check if do proto files dump also....

closed 2023-10-12 by samulimaa