polarofficial/polar-ble-sdk #64: Problem with RxAtomic when trying to run iOS acc demo

joezein, 2020-03-10:
```
dyld: Library not loaded: @rpath/RxAtomic.framework/RxAtomic
  Referenced from: /private/var/containers/Bundle/Application/B9348436-9CDB-49A5-A180-33C7C58D2DBD/PolarSDK-ACC-Demo.app/PolarSDK-ACC-Demo
  Reason: no suitable image found.  Did find:
	/private/var/containers/Bundle/Application/B9348436-9CDB-49A5-A180-33C7C58D2DBD/PolarSDK-ACC-Demo.app/Frameworks/RxAtomic.framework/RxAtomic: code signature invalid for '/private/var/containers/Bundle/Application/B9348436-9CDB-49A5-A180-33C7C58D2DBD/PolarSDK-ACC-Demo.app/Frameworks/RxAtomic.framework/RxAtomic' 
```

Hi, I am getting this error every time I try to run this on my device. App on my phone stays stuck on LaunchScreen. 

Can you please help me make it work?

I am trying to run the iOS Acc demo

joezein, 2020-03-10:
[image]

This is what I get

erkki-silvola, 2020-03-10:
Hi,

I have not checked the ACC demo for a long time, so it might be outdated... I will check it this week, mean while try the exampleApp which should work just fine

joezein, 2020-03-10:
I  still get the  same issue for example. Failed to register bundle+ No profile

erkki-silvola, 2020-03-11:
Try to change the bundle identifier

pajama, 2020-03-27:
I was able to get it to build by removing the rxatomic framework on the general tab. RxSwift is the replacement to atomic.

[image]

I then had to add the bluetooth usage description to the info.plist for it to not crash on load.

However for any of the demos/examples that support ACC I get the error 
`PolarBleSdk.InvalidArgument`

pajama, 2020-03-27:
I was able to get it to work by updating the firmware with the official polar app.

kkizlaitis, 2020-04-25:
@pajama is right. Update the frameworks, remove `RxAtomic.framework` and also make sure that _Embed & Sign_ is selected as in the screenshot.

closed 2023-09-08 by jimmyzumthurm