polarofficial/polar-ble-sdk #65: How to Get RR value in HeartRateMeasurement

ltvuongqn88, 2020-03-14:
Dear Polar team,

Please help me get RR value from byte[] data received HeartRateMeasurementCharacteristic.ValueChanged.

I can get HR value by method:

public static ushort ParseHeartRateValue(byte[] data)
        {
            // Heart Rate profile defined flag values
            const byte heartRateValueFormat = 0x01;

            byte flags = data[0];
            bool isHeartRateValueSizeLong = ((flags & heartRateValueFormat) != 0);

            if (isHeartRateValueSizeLong)
            {
                return BitConverter.ToUInt16(data, 1);
            }
            else
            {
                return data[1];
            }
        }

erkki-silvola, 2020-03-20:
Hi, there is source code dump you check from there

jimmyzumthurm, 2023-09-08:
Hi, as @erkki-silvola  said, you can find RR data in PolarHrData, and there is flag to check if data is available or not : https://github.com/polarofficial/polar-ble-sdk/blob/fdb5bd0c485eecbfc857708f8f9c65efd02052ae/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.kt#L4

closed 2023-09-08 by jimmyzumthurm