polarofficial/polar-ble-sdk #66: OH1: Enabling PPI lowers HR sample frequency to 5 Hz

erikboto, 2020-03-18:
It seems like the HR sampling rate is slowed down to 5 Hz when the PPI measurement it enabled. I still get updates on the HR characteristic at 1 Hz, but the value only changes in sync with the PPI value updates which happen at 5 Hz.

Is this a bug, or maybe due to limited processing power or something like that?

erikboto, 2022-10-17:
You will get a PPI value and a corresponding estimate of its accuracy for every heart beat. If I recall correctly they are sent in batches from the OH-1 approximately every 5th second. But that doesn't really matter if you just want to be able to calculate HRV as you should still get all values.

You need to be rather still when using PPI values though, otherwise they will not be very accurate.

jimmyzumthurm, 2023-09-08:
Hi @erikboto , this answer might come in late, but I have explanation for this behavior. Basically, when PPI data is calculated, it is a separate algorithm that is running from the main OHR one. The main OHR algorithm calculates accurate HR data every second with motion cancellation by using accelerometer data. PPI are calculated over a 5 second window, meaning that each 5 seconds there is an update on the HR value, HR is more of a side feature of the PPI algorithm. Also note that this algorithm is very sensitive to movement ("blocker bit" of PPI data packet indicates that movement was present and that data should not be trusted) compared to the OHR one that is used for exercise.

Unfortunately this is limitation by the algorithms that were developed for these older products and we are not planning on maintaining those.

closed 2023-09-08 by jimmyzumthurm