polarofficial/polar-ble-sdk #68: HCG-HR Demo can't connect to Device

frankzhouhr, 2020-04-14:
Hi, I ran the Android sample code and I was able to see HR and ECG logs. But when I built and ran the demo app, when I click connect ECG, I saw "connecting to Device XXXXX" but no further prompts. There was an unhandled exception in callback showing below, 

    W/BluetoothGatt: Unhandled exception in callback
    java.lang.RuntimeException: Can't toast on a thread that has not called Looper.prepare()
    ...
    at com.polar.polarsdkecghrdemo.ECGActivity$1.deviceConnected(ECGActivity.java:70)
    ...

I used Pixel 3a and the firmware version for H10 is 3.4.5. Can you help me figure out why?

Thank you!

erkki-silvola, 2020-04-15:
Hi,

Fixed in master

frankzhouhr, 2020-04-15:
Working now, thanks for the update!

closed 2020-04-15 by frankzhouhr