polarofficial/polar-ble-sdk #69: ACC error: PolarBleSdk.NotificationNotEnabled

pajama, 2020-04-16:
Device connects and reports proper device id, accFeatureReady callback fires. But I still get
ACC error: PolarBleSdk.NotificationNotEnabled 

```
public func accFeatureReady(_ identifier: String) {
        NSLog("ACC READY")
        tryAccToggle()
    }

func tryAccToggle() {
        if accToggle == nil {
            accToggle = api.requestAccSettings(deviceId).asObservable().flatMap({ (settings) -> Observable<PolarAccData> in
                NSLog("settings: \(settings.settings)")
                return self.api.startAccStreaming(self.deviceId, settings: settings.maxSettings())
            }).observeOn(MainScheduler.instance).subscribe{ e in
                switch e {
                case .next(let data):
                    for item in data.samples {
                        let accelString = "\(item.x),\(item.y),\(item.z)"
                        NSLog(accelString)
                        self.accelStreamHandler?.eventSink!(accelString)
                    }
                case .error(let err):
                    NSLog("ACC error: \(err)")
                    self.accToggle = nil
                case .completed:
                    break
                }
            }
        } else {
            accToggle?.dispose()
            accToggle = nil
        }
    }
    
```

pajama, 2020-04-16:
I was able to get it to work by adding a 2 second delay to calling tryAccToggle, after the ACC ready feature fires. Is there another callback for when it is actually ready?

It isn't failing anymore at least, but the data streaming still doesn't start immediately after subscribing. It's pretty unpredictable when it actually starts, sometimes it can be up to a minute later.

Thanks for any help you can offer!

erkki-silvola, 2020-04-16:
Hi,

Fixed in master

pajama, 2020-04-16:
Thanks for working on this!

Now I'm getting
settings: [PolarBleSdk.PolarSensorSetting.SettingType.range: Set([8]), PolarBleSdk.PolarSensorSetting.SettingType.sampleRate: Set([50]), PolarBleSdk.PolarSensorSetting.SettingType.resolution: Set([16])]
2020-04-16 10:47:01.095086-0700 Runner[458:34746] ACC error: DeviceError(localizedDescription: "controlPointRequestFailed(errorCode: 6)")

pajama, 2020-04-16:
Sorry, disregard my last comment. Looks like it was still streaming to another app in the background. Shutting that down and restarting the sensor fixed it. Looks like stream can be toggled after the acc ready callback fires now! Thanks!

closed 2020-04-16 by pajama