polarofficial/polar-ble-sdk #73: Class X is implemented in both X and X (protobuf classes)

simonbengtsson, 2020-04-20:
We are using Firebase with our app which depends on protobuf. Following the integration guide and just adding the PolarSDK and ReactiveX frameworks we got a lot of warnings such as this:

> Class X is implemented in both /.../PolarBleSdk.framework/PolarBleSdk and /.../protobuf.framework/protobuf. One of the two will be used. Which one is undefined.

It seems there is a protobuf version bundled with the framework? Is it possible use the protobuf version we already have instead? 

Would be amazing to get official Cocoa Pods support as well. Or am I just missing something?

<details><summary>Complete warnings</summary>
<p>

```
objc[2031]: Class GPBAnyRoot is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087be960) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36cb0). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBAny is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087be988) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36cd8). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBApiRoot is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087bea00) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36d50). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBApi is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087bea28) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36d78). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBMethod is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087bea78) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36dc8). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBMixin is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087beac8) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36e18). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBDurationRoot is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c0440) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36e90). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBDuration is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c0468) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36eb8). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBEmptyRoot is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c1d90) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36f30). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBEmpty is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c1db8) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36f58). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBFieldMaskRoot is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c1340) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36fd0). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBFieldMask is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c1368) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b36ff8). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBInt32Array is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c3168) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b37048). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBUInt32Array is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c31b8) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b37098). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBInt64Array is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c3208) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b370e8). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBUInt64Array is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c3258) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b37138). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBFloatArray is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c32a8) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b37188). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBDoubleArray is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c32f8) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b371d8). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBBoolArray is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c3348) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b37228). One of the two will be used. Which one is undefined.
[... cut ...] uf (0x107b38f38). One of the two will be used. Which one is undefined.
objc[2031]: Class GPBBytesValue is implemented in both /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/PolarBleSdk.framework/PolarBleSdk (0x1087c1d18) and /private/var/containers/Bundle/Application/2AC4A745-5EF6-47D3-AB6D-1B7F6282BB70/Equilab.app/Frameworks/protobuf.framework/protobuf (0x107b38f88). One of the two will be used. Which one is undefined.
```

</p>
</details>

erkki-silvola, 2020-04-20:
Hi,

I will check the protobuf separation from the sdk, meanwhile you can configure pod to exclude the protobuf framework. 

If this will be done properly only proto files would be distributed as protoc and protobuf needs to be the same version.

erkki-silvola, 2020-04-28:
Hi,

There is a branch swift_protobuf_support where I switched protobuf dep to apple swift version

simonbengtsson, 2020-04-28:
I experimented with trying to create a pod without the protobuf, but didn't get it to work. Cool with the swift protobuf. I guess not optimal but would at least solve the duplicate classes issue for us I guess. I'll check it out 👌

closed 2023-10-12 by samulimaa