polarofficial/polar-ble-sdk #75: Accelerometer stream starts inconsistently

pajama, 2020-04-20:
What are the requirements for getting the accelerometer stream to start? 

Occasionally the first launch will start streaming events immediately after the ready callback has fired. 

On subsequent launches, it can be very inconsistent. Despite the ready callback always firing, it can take a minute or longer for the events to start streaming.

erkki-silvola, 2020-04-21:
Hi,

Device will request faster connection parameters, and depending on the phone etc.. it may take take some time to get desired parameters in use

outsbook, 2020-06-10:
You need to start accelerometer stream after "**accelerometerFeatureReady**" callback

```
@Override
public void accelerometerFeatureReady(String identifier) {
      //start accelerometer stream here 
}
```

pajama, 2020-06-11:
thanks @outsbook. I am already doing that.

JOikarinen, 2021-02-05:
@pajama could you still recall what is the phone model you faced the problem? Which Polar model you used when noticing the problem,  Polar H10? I could try to repeat the issue.

jimmyzumthurm, 2023-09-08:
Closing this as this has been stale for over 3 years

closed 2023-09-08 by jimmyzumthurm