polarofficial/polar-ble-sdk #76: ECG Demo builds failing

dhuszaroakley, 2020-04-22:
Hi, I'm having trouble building the ECG Demo application. I forked the repository sometime yesterday, but only just noticed that the master branch has been receiving patches to update the libraries used. I'm wondering if I forked the repository while it was in the process of being updated as my builds are failing due to "cannot access class file" errors. 

I don't have a lot of experience in android development, but from what I can tell, the compiler is looking for "io.reactivex.rxjava3" class files. Part of RxJava version 3. I could be barking up the wrong tree with this, but the ECG Demo doesn't seem to require v3, only V2 as dependencies, at least on my fork of the repository.

A little guidance would be greatly appreciated 😇

erkki-silvola, 2020-04-23:
Hi, 

As I updated few days ago android sdk to rx3, and yesterday I updated the ecg demo too, so you need to update the ecg demo app sources

dhuszaroakley, 2020-04-23:
Thanks! Just earlier I fetched the latest version of the repository and it built with no problems. I think I just was unfortunate to fork just after the SDK libs were updated but before the ECG Demo was brought up to date. Nevertheless it's all working now, I just didn't have the experience to know what the actual issue was. Thanks again.

closed 2020-04-23 by dhuszaroakley