polarofficial/polar-ble-sdk #77: Could I choose a device to connect when there are 2 devices

capmasa, 2020-04-30:
Hi,
I developing fitness app getting bpm by polar devices.Then, sum users enjoying fitness with partners having each polar devices and apps.

Could I make users choose which device connect to apps?

Thanks

capmasa, 2020-05-02:
Sorry I founded it was simple way.
1. searchDevices
2. choose device id using something(I used alertview)
3. call startListenForPolarHrBroadcasts with chosen id

thanks

closed 2020-05-02 by capmasa

jams032, 2020-05-27:
@capmasa  , say you have 5 polar devices to be discovered. We have to discover first then request for connection. Are you able discover those devices/sensors using searchDevices() method before calling 

try api.connectToDevice(deviceID) ? 

On which callback method , you get response for the discovered devices ? 

Could you be bit elaborate here ? I did not able to get any response from searchDevices() method, so that i used native BLE scanner and then discover to connect with Polar API.