polarofficial/polar-ble-sdk #78: Always shows H10 battery is 10

KennethEvans, 2020-05-02:
The battery level is always 10 for my H10.  This is unlikely as it is over a year old and sees at least 3 hours a week usage.

I have another BLE application completely unrelated to the Polar SDK.  It also shows 10 for the battery level.  It works as expected on other devices, and I have seen several batteries run down.  The H10 hasn't shown any decline.  I assume this is owing to the fact that BLE using the usual Heart Rate service isn't getting the right value from the device.

On the other hand Polar Beat is now saying it is low.

So three questions:
Why is the Polar SDK (and my other app) giving the wrong value?
What is Beat doing differently that it gets the right value.
Can this be done with the Polar SDK?

Thanks.

KennethEvans, 2020-05-03:
Sorry, that was a mental glitch on my part.  It was saying 10, not 100.  Yes, it was nearly dead.  It stopped responding to the Polar SDK app, but still responded to the other BLE app. After putting in a new battery it was ok (and I noticed it should be saying 100).

Part of the reason I was mislead is because it read 100 for maybe 6 months after getting it, and I had the idea in the back of my head it always read the same.  I have some data however, that indicates it was down to 60 by January this year, though I didn't notice.  (I got it in January last year.)  In any event it lasted a long time, especially since I had it on for hours at a time when debugging about this time last year.

I apologize.  This can be closed.  (Deleted would be even better, since it is embarrassing.)

JOikarinen, 2021-02-05:
@KennethEvans no worries. Sorry for very late response, closing this now.

closed 2021-02-05 by JOikarinen