polarofficial/polar-ble-sdk #82: connectToDevice() function not working when app in background

outsbook, 2020-06-06:
**Device: Polar NH10**
I am using `api.connectToDevice(DEVICE_ID); `to connect the device within android foreground service.
In my app here is functionality if somehow the polar device disconnected then we try to connect the device with `api.connectToDevice(DEVICE_ID);` function.

**In Android Marshmallow**
It's working fine for both foreground mode and background mode.

**In Android 10**
It's working fine when the app in foreground mode but when the app is in background mode it's not working.

For your information I use `api.backgroundEntered();` and `api.foregroundEntered();`

Could anyone please help me to solved this issue?

_Thanks for reading..._

kcbaruri, 2020-06-07:
Did you use as below?
 @Override
    protected void onPause() {
        super.onPause();
        polarBleApi.backgroundEntered();
    }

    @Override
    protected void onResume() {
        super.onResume();
        polarBleApi.foregroundEntered();
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        polarBleApi.shutDown();
    }

outsbook, 2020-06-07:
@kcbaruri Yes, I use all 3 functions as same as your example

outsbook, 2020-06-07:
Finally I found the solution, it was location permission issue.
Need Background Location permission
`<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />`

closed 2020-06-07 by outsbook

sach16795, 2021-08-03:
@JavaDeveloper08 Were you able to use the service to record values? Right now I can store the values in the main activity. But it will not do the same if the app is killed. I think a background service is needed. But my efforts to implement a service is failing due to some issue in requestStreamSettings method. 

If you have solved this issue please let me know if you can help me out. Thanks in advance.