polarofficial/polar-ble-sdk #84: Could not read characteristics

darkzbaron, 2020-06-09:
HI there, I am trying to get HR data with Bleak and I get the following message

``
[Service] 0000180d-0000-1000-8000-00805f9b34fb: Heart Rate
        [Characteristic] 00002a37-0000-1000-8000-00805f9b34fb: (notify) | Name: , Value: None
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 17) | Value: b'\x00\x00'
``

I have tried many ways to get HR with no success. Any help?

Thanks in advance

outsbook, 2020-06-10:
Did you use 
`PolarBleApi api = PolarBleApiDefaultImpl.defaultImplementation(this, PolarBleApi.FEATURE_HR);`

**PolarBleApi.FEATURE_HR** only for read heart rate

If you done that then you get the heart rate and other data in 
  ```
    @Override
    public void hrNotificationReceived(String identifier, PolarHrData data) {
        Log.d("MyApp","HR: " + data.hr);
    }
```

darkzbaron, 2020-06-11:
Issue was started by using read and not notify on sensor

closed 2020-06-11 by darkzbaron

darkzbaron, 2020-06-25:
Any help on the PPI I can't seem to get the PPI intervals using the following code 

`` res["rr_interval"] = ((byte0 >> 4) & 1) == 1 ``

It works for the H10 but not the OH1. Am I missing something?

Thanks very much in advance

outsbook, 2020-06-25:
@darkzbaron OH1 not send rr_interval value

darkzbaron, 2020-06-25:
Yes, I know I am talking about PPI