polarofficial/polar-ble-sdk #87: Start ppg error iOS

somparkash, 2020-07-07:
Dear Polar

after getting "ohrPPGFeatureReady" delegate called then I started "startOhrPPGStreaming" but PPG streaming won't able to start and getting following error:

"start ppg error: DeviceError(localizedDescription: "controlPointRequestFailed(errorCode: 6)")"

Device Information:
Software Revision String : 2.0.11
System ID : k þÿ 
batteryLevelReceived 90

Could anyone please help me with this.

Thank you

msippo, 2020-07-08:
Hi,

Have you checked that you have paired OH1 with your iOS device? You should see Polar OH1 in the Bluetooth Settings device list. Bluetooth Errorcode 6 means PIN or KEY missing. You can read the list of the error codes from BLUETOOTH CORE SPECIFICATION  Vol 1, Part F.

Br,
Matti from Polar

somparkash, 2020-07-08:
Hi @msippo,

Thank you for your response. Yes OH1 is paired with my app and I successfully get the heart rate. But some times when I tried to start PPH get this error. Then I restart my polar device and It worked. Is this device specific issue or I missed something in the code.

Thanks,
Som

JOikarinen, 2021-01-15:
@somparkash I double check the code and error reason. The errorCode: 6 actually means "Invalid state" returned from the OH1. Could you describe exactly the steps in which conditions you get that response? Please feel free to reopen the issue if you face the problem again.

closed 2021-01-15 by JOikarinen