polarofficial/polar-ble-sdk #90: Android sdk integration into flutter fails at runtime: `Failed resolution of: Lio/reactivex/rxjava3/android/schedulers/AndroidSchedulers;`

RickVM, 2020-08-13:
We're trying to setup the [polar-ble-sdk][1] for usage in our flutter app so that we can use the internal heart rate recording from the H10. We do this in flutter via platform channels.

After installing the dependencies (2 .aar files from polar) we get a runtime error where one of the dependent classes cannot be found, despite being in the sdk.
This problem does not occur if we load the sdk in a native android project without flutter.
Since we took all 'expected' steps we'd think this should work in flutter.

`[ERROR:flutter/shell/platform/android/platform_view_android_jni_impl.cc(43)] java.lang.NoClassDefFoundError: Failed resolution of: Lio/reactivex/rxjava3/android/schedulers/AndroidSchedulers;`

**Things we've tried**

1. Analyze if multidex is required -> first project was over 65k methods
2. Add  / verify multidex is supported by both using a minimal android sdk > 21 and manually enabling it
3. Creating a MVP project with java rather than kotlin support where method references were about 50K (no multidex required)
4. Upgrading to flutter v1.20
5. Removing .aar files to see if they were even used; this results in build time error. Suggesting they are found. Also, you'd expect they are integrated since we can import the dependencies at runtime. Just not use them..
6. Ensured androixX is enabled together with jetifier

**How to reproduce**

Follow the steps listed in the [polar-ble-sdk:][1]

1. in android/src/app/libs add polar-ble-sdk.aar and polar-protobuf-release.aar from their [github][2]
2. in build .gradle add depdencies
```
dependencies {
    implementation files('libs/polar-ble-sdk.aar')
//    // Only needed if FEATURE_POLAR_FILE_TRANSFER used
    implementation files('libs/polar-protobuf-release.aar')
//    // Only needed if FEATURE_POLAR_FILE_TRANSFER used
    implementation group: 'commons-io', name: 'commons-io', version: '2.4'
//    // Only needed if FEATURE_POLAR_FILE_TRANSFER used
    implementation 'com.google.protobuf:protobuf-java:3.1.0'
    implementation 'io.reactivex.rxjava2:rxjava:2.1.5'
    implementation 'io.reactivex.rxjava2:rxandroid:2.0.1'
}
```

3. Setup android manifest to get permissions

```
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```

4. Add native imports & code 

```
package com.example.flutter_app

import androidx.annotation.NonNull
import io.flutter.embedding.android.FlutterActivity
import io.flutter.embedding.engine.FlutterEngine
import io.flutter.plugin.common.MethodChannel

import android.content.Context
import android.content.ContextWrapper
import android.content.Intent
import android.content.IntentFilter
import android.os.BatteryManager
import android.os.Build.VERSION
import android.os.Build.VERSION_CODES
import android.util.Log

import polar.com.sdk.api.PolarBleApi;
import polar.com.sdk.api.PolarBleApiDefaultImpl;

// also tried with these imports later, according to sdk

import io.reactivex.CompletableObserver;
[... 131 lines cut ...]
E/flutter (28735): 	at com.example.flutter_app.MainActivity.initApi(MainActivity.kt:62)
E/flutter (28735): 	at com.example.flutter_app.MainActivity.access$initApi(MainActivity.kt:41)
E/flutter (28735): 	at com.example.flutter_app.MainActivity$configureFlutterEngine$1.onMethodCall(MainActivity.kt:51)
E/flutter (28735): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:230)
E/flutter (28735): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
E/flutter (28735): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:692)
E/flutter (28735): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (28735): 	at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter (28735): 	at android.os.Looper.loop(Looper.java:174)
E/flutter (28735): 	at android.app.ActivityThread.main(ActivityThread.java:7682)
E/flutter (28735): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (28735): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:516)
E/flutter (28735): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
E/flutter (28735): Caused by: java.lang.ClassNotFoundException: Didn't find class "io.reactivex.rxjava3.android.schedulers.AndroidSchedulers" on path: DexPathList[[zip file "/data/app/com.example.flutter_app-BsARJP-hOr4YHOh1VaUPIw==/base.apk"],nativeLibraryDirectories=[/data/app/com.example.flutter_app-BsARJP-hOr4YHOh1VaUPIw==/lib/arm64, /data/app/com.example.flutter_app-BsARJP-hOr4YHOh1VaUPIw==/base.apk!/lib/arm64-v8a, /system/lib64, /system/product/lib64]]
E/flutter (28735): 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:196)
E/flutter (28735): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
E/flutter (28735): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
E/flutter (28735): 	... 17 more
E/flutter (28735): 
F/flutter (28735): [FATAL:flutter/shell/platform/android/platform_view_android_jni_impl.cc(942)] Check failed: CheckException(env). 
F/libc    (28735): Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 28735 (ple.flutter_app), pid 28735 (ple.flutter_app)
```

  [1]: https://github.com/polarofficial/polar-ble-sdk
  [2]: https://github.com/polarofficial/polar-ble-sdk/tree/master/polar-sdk-android/libs

msippo, 2020-08-14:
Hi,

We have updated to RX3.
Change build.gradle lines about reactivex to
    implementation 'io.reactivex.rxjava3:rxjava:3.0.0'
    implementation 'io.reactivex.rxjava3:rxandroid:3.0.0'

Check from our latest examples how this affects to the code.

Br,
Matti

RickVM, 2020-08-14:
That fixed it. Thanks! 
Might want to update the instructions on readme.md in the root folder, its still referencing RX2
If you'd like i can make a pull request

JOikarinen, 2021-01-12:
@RickVM Readme is (finally) updated.

closed 2021-01-12 by JOikarinen